raspiblitz/raspiblitz #2262: Test rotated Touchscreen

rootzoll, 2021-05-02:
Had a report with the v1.7 that the touchscreen is not running anymore in rotate. Check and fix for v1.7.1

Natzn123, 2021-05-02:
I want to add my experience with this issue:
1. deactivate lcd rotate option and activate touchscreen option in one step
2. after reboot both happened, but touchscreen was not working correctly
3. I did a Touchscreen Calibration several times, but doesn't work and I recognized the following behaviour:
[image]
               | Touch | Pointer
touching | top left | bottom right
touching | top right | top right
touching | bottom left | bottom left
touching | bottom right | top left
sliding | top left --> top right | bottom right --> top right
sliding | top left --> bottom left | bottom right --> bottem left
sliding | top right --> top left | top right --> bottom right
sliding | top right --> bottom right | top right --> top left
sliding | bottom left --> top left | bottom left --> bottom right
sliding | bottom left --> bottom right | bottom left --> top left
sliding | bottom right --> top right | top left --> top right
sliding | bottom right --> bottom left | top left --> bottom left

4. I decided to activate lcd rotate again
5. it doesn't work without touchscreen calibration, but I recognized, that it is now the following situation: touching top left corner of rotated screen (so in scenario 3, it was the bottom right corner) the pointer is located at the top right corner, no further tests done here.
6. doing touchscreen calibration and afterwards it worked with setup rotate LCD activ.
7. I decided to deactivate rotate LCD again, but same scenario as described under point 3
8. I thought, that touchscreen must be activated while in "not rotate LCD mode", so deactivate touchscreen and afterwards activate touchscreen again while rotate lcd deactivated, but same scenario like under point 3.

rootzoll, 2021-09-20:
Moved to v1.8.0 because we will do Touchscreen related changes anyway.