raspiblitz/raspiblitz #2264: SCB not send to DropBox when new channel backup is generated.

mrsieb, 2021-05-02:
I should mention i am still on 1.7 Dev RC2 

I set up DropBoxBackup and the db.file is send to dropbox right away. But get not updatet after new channels are opened.

```
 sudo journalctl -f -u background
-- Logs begin at Sun 2021-05-02 06:08:50 BST. --
May 02 12:29:14 raspberrypi _background.sh[1370]: Blockchain Sync Monitoring: peers=20
May 02 12:29:14 raspberrypi sudo[23751]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/ls /mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup
May 02 12:29:14 raspberrypi sudo[23751]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 02 12:29:14 raspberrypi sudo[23751]: pam_unix(sudo:session): session closed for user root
May 02 12:29:14 raspberrypi sudo[23756]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/md5sum /mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup
May 02 12:29:14 raspberrypi sudo[23756]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 02 12:29:14 raspberrypi sudo[23756]: pam_unix(sudo:session): session closed for user root
May 02 12:29:14 raspberrypi sudo[23761]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/md5sum /home/admin/backups/scb/channel.backup
May 02 12:29:14 raspberrypi sudo[23761]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 02 12:29:14 raspberrypi sudo[23761]: pam_unix(sudo:session): session closed for user root
May 02 12:30:15 raspberrypi sudo[26055]:     root : TTY=unknown ; PWD=/ ; USER=admin ; COMMAND=/home/admin/config.scripts/network.monitor.sh peer-status
May 02 12:30:15 raspberrypi sudo[26055]: pam_unix(sudo:session): session opened for user admin by (uid=0)
May 02 12:30:15 raspberrypi sudo[26055]: pam_unix(sudo:session): session closed for user admin
May 02 12:30:15 raspberrypi _background.sh[1370]: Blockchain Sync Monitoring: peers=20
May 02 12:30:15 raspberrypi sudo[26062]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/ls /mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup
May 02 12:30:15 raspberrypi sudo[26062]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 02 12:30:15 raspberrypi sudo[26062]: pam_unix(sudo:session): session closed for user root
May 02 12:30:15 raspberrypi sudo[26067]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/md5sum /mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup
May 02 12:30:15 raspberrypi sudo[26067]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 02 12:30:15 raspberrypi sudo[26067]: pam_unix(sudo:session): session closed for user root
May 02 12:30:15 raspberrypi sudo[26075]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/md5sum /home/admin/backups/scb/channel.backup
May 02 12:30:15 raspberrypi sudo[26075]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 02 12:30:15 raspberrypi sudo[26075]: pam_unix(sudo:session): session closed for user root
```
I see the backup gets triggerd  but on DropBox its still 3 days old. Are there more related logs? Maybe this is just a access issue with dropbox ?

nyxnor, 2021-05-07:
Can you try to upload with this command:
```
bash /home/admin/config.scripts/dropbox.upload.sh upload [AUTHTOKEN] [FILEPATH]
```

mrsieb, 2021-05-07:
Got this output

`dropbox.upload.sh: 16: dropbox.upload.sh: source: not found
dropbox.upload.sh: 20: [: 9CXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXSGy7ho: unexpected operator
dropbox.upload.sh: 45: [: 9CXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXSGy7ho: unexpected operator
dropbox.upload.sh: 50: [: 9CXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXSGy7ho: unexpected operator
dropbox.upload.sh: 70: [: 9CXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXSGy7ho: unexpected operator
err='unkown mode'
`

nyxnor, 2021-05-07:
Can you share the command you typed? I think you typed upload wrong. Also, maybe try to set it `off` then `on` again

mrsieb, 2021-05-08:
well ok i used sh instead of bash to call the script but now i got this:

`# Upload happened but is size zero
upload=0
err='size zero'
errMore='name lnXXXXXXX57 pathlower lndbacxxxxxxxxxx3857 pathdisplay lndbacxxxxxx3857 id idUqbxxxxxxxxAAAAVsA clientmodified 20210508T083518Z servermodified 20210508T083518Z rev 015c1cd6cxxxxxxxxxxxxxx360a4e20 size 0 isdownloadable true contenthash e3b0c44298fc1cxxxxxxxxxxxxxxxxxxxxxxxxxxxxca495991b7852b855'
`

mrsieb, 2021-05-08:
Well a new folder was created on DB but no file uploaded

nyxnor, 2021-05-08:
There is no data in that file. 

Check where it is with:
```
sudo find / -name channel.backup
```
The uploads should be automatically done, so start over.
```
bash /home/admin/config.scripts/dropbox.upload.sh off
```
Do these steps again: https://github.com/rootzoll/raspiblitz#a-dropbox-backup-target
```
bash /home/admin/config.scripts/dropbox.upload.sh on [AUTHTOKEN]
```

mrsieb, 2021-05-10:
sudo find / -name channel.backup
/boot/channel.backup
/home/admin/backups/scb/channel.backup
/mnt/hdd/app-data/lnd/data/chain/chain/bitcoin/mainnet/channel.backup
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup
/mnt/hdd/backup_lnd/data/chain/bitcoin/mainnet/channel.backup
find: ‘/proc/549’: No such file or directory
find: ‘/proc/550’: No such file or directory
find: ‘/proc/551’: No such file or directory
find: ‘/proc/562’: No such file or directory
find: ‘/proc/563’: No such file or directory
find: ‘/proc/564’: No such file or directory

Well i will try to set up again, and i am aware of the step to seht the write permission firts.

rootzoll, 2021-05-18:
I am not sure thats the cause in this particular case ... but will add this feature for re-test for v1.7.1 release.

michaelg-au, 2021-06-19:
I'm not sure if this is the answer, but I had a similar problem with the scb upload to dropbox working temporarily and then stopping. I also noticed a few people on the Raspiblitz Telegram commenting the same. If you exactly follow the instructions provided at https://github.com/rootzoll/raspiblitz#a-dropbox-backup-target - then you will leave the access token expiration as 'short lived'. I changed it to 'no expiration' (as it was set for previous dropbox tokens on blitz instances I had last year). I'm hoping this fixes things for me. Maybe addresses the above (was going to raise a new Issue that the instructions might need an update, but saw this Issue).

rootzoll, 2021-07-01:
@michaelg-au thanks - will look for the "no expiration" option, could be that this one is new.

michaelg-au, 2021-07-02:
[image]

Yes - this is an extract of the email from Dropbox

happyluckbox, 2021-07-05:
They said the no expiration token will be deprecated in the future. Not sure what this means?

rootzoll, 2021-08-17:
This one is still on the fix list for v1.7.1 and will be addressed after RC1

rootzoll, 2021-09-20:
Yeah the expiration of the token seems to be the breaking point - updated the documentation to have the token with "no expiration". Its working still at the moment - but DropBox deprecation is coming on 30th September. Thats why we will also deprecate the DropBox Backup with v1.7.1 and will recommend people changing to a NextCloud Backup which is introduced with v1.7.1.

TODO: Update Documentation (remove DropBox, add Nextcloud)

rootzoll, 2021-10-04:
Documentaion for Nextcloud updated.

closed 2021-10-04 by rootzoll