raspiblitz/raspiblitz #2267: Feature request - Taproot enforcement

Tracachang, 2021-05-03:
Hi,

Since taproot is going to be activated in a few months, it is really important that users update their client. Bitcoin core is using speedy trial while other clients are enforcing it by bip8 (lot=true). 

If miners do their work and ugrade, all will be fine no matter which software we use since taproot will be activated.  But if miners not signal at least 90% there will be the risk of split/fork, and those using speedy trial risk to loose some funds while their nodes will accept both blocks (with and without taproot) and nodes that enforce taproot will reject blocks not signaling taproot.

If lot=true is the longest chain,  will be accepted by all nodes (making some past tx with no taproot invalid).

I do believe the safest bet is to enforce taproot and since there's consensus with taproot activation so let's make it happen.

Nodes operators must think and make a decision, that is why I am asking  the possibility to choose if  they decide or not (and be aware of risks) in a easy way for non technical users. 

Has the possibility of using as default bitcoin v0.21.0-taproot0.1 been considered? 

Thanks

rootzoll, 2021-05-03:
Because the RaspiBlitz users are already using SSH for login its very easy to excute manual update scripts @openoms did that in the past. So there is always that fallback option that is available even if we from the dev team side do not react in all detail or fast enough.

An update of bitcoin core to v0.21.1 is alraedy possible with the latest realease v1.7.0 if you go `MAINMENU > UPDATE > BITCOIN > CUSTOM > enter 0.21.1` ... this works for all releases that get available from the bitcoin core github. There is no SSH UI for updating to bitcoin core releases outside of official bitcoin core github yet ... but again with all our users having the terminal at their fingertips this always can be done if necessary.

openoms, 2021-05-03:
I don't think we should default to a non Bitcoin Core release, but everyone is free to install an alternative client.

https://stadicus.github.io/RaspiBolt/raspibolt_30_bitcoin.html#installation describes the manual process of binary download, verification and installation.

Will consider making a PR to offer the UASF client as an update option if the Speedy Trial fails to have Taproot locked in.

Tracachang, 2021-05-03:
Thanks for your answers, sure everyone is free to install an alternative client, but how many people do it?

What I am trying to say is that people who run node solutions like raspiblitz and the rest of competitors... some they do because it is nice and easy, but most of them it is because they are not technical and they want a ready 2 go solution.

What is happening right now with taproot seems to not be taken seriously enough, most of people say "let's wait until november and if miners do not upgrade then we can try a UASF" which is not true and can have real consequences.

Since there is consensus and miners have enough time to upgrade there is no reason to not do it, so miners can block taproot activation just by refusing to signal. It does not makes sense at all. 

But thanks to some devs, we have UASF clients running, so no matter what, taproot will be activated. 

> LOT=false nodes, in contrast, would accept the LOT=true chain if it ever became the longer chain. They’d then abandon “their own” LOT=false chain, and lose any coins they received or mined on it since the split. In short: **LOT=false nodes and miners would risk losing money**. Therefore, they’d have a strong incentive to just join the LOT=true chain before this can happen. (Especially since they, presumably, don’t object to the Taproot upgrade itself.)` can read the full article https://bitcoinmagazine.com/technical/lottrue-or-lotfalse-this-is-the-last-hurdle-before-taproot-activation

I insist in the possibility since I think it is really important for users to choose, because updating to ST, not doing nothing or updating to UASF client all have consequences, and most of node operators use solutions like Raspiblitz, so giving the choice (as easier as possible) I do think it is the correct way.

rootzoll, 2021-05-03:
RaspiBlitz is an OpenSource project .. if you want to provide an "easy solution of choice" to the community please make a PR that adds that choice to ` MAINMENU > UPDATE > BITCOIN` menu - this is file your looking for: https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/config.scripts/bitcoin.update.sh

openoms, 2021-05-03:
@Tracachang the Speedy Trial ends before November, less than 3 months from now.

The earliest activation of Taproot is 2021 November if it locks in during ST.
The UASF client will enforce Taproot only from 2022 October.

It does absolutely nothing for the network consensus if one runs the UASF client now and it might not make any difference either if Taproot locks in during ST.

Installling an alternative client come with other questions as well. BItcoin Core is the most well reviewed release and afaik there is only one well known bitcoin developer is signing the UASF client.
One who is not able to install manually is surely hasn't reviewed the code differences between the clients either.
This is not the time to force an alternative client and I don't think it needs to be easily offered yet.
This next ~3 months is the time for the miners to show support and possibly facilitate the move. If it doesn't happen we can and will take further steps.

Tracachang, 2021-05-04:
Thanks for your answer @openoms.

I'll close the feature request since you are right:

`This is not the time to force an alternative client and I don't think it needs to be easily offered yet.
This next ~3 months is the time for the miners to show support and possibly facilitate the move. If it doesn't happen we can and will take further steps.`

closed 2021-05-04 by Tracachang