raspiblitz/raspiblitz #2269: Can't connect to my node from Fully Noded

xpsx, 2021-05-03:
Good evening,

I recently updated my raspiblitz from 1.6.3 to 1.7.0
and now I can't connect to my node from (Fully Noded)
the node is really slow when it I want to view the info.
here is the debug log:

https://termbin.com/yjup

openoms, 2021-05-03:
Try using the CONNECT menu to connect Fully Noded.
The hidden service is not exposed by default any more.

xpsx, 2021-05-04:
I already tried it, it gives me error connection was lost, after timeout

openoms, 2021-05-04:
Running this has helped people.before recreating the Hidden Service:
```
/home/admin/config.scripts/internet.hiddenservice.sh bitcoin8332 8332 8332
```
restart Tor or the whole node to rebuild the Tor circuits.

xpsx, 2021-05-04:
> Running this has helped people.before recreating the Hidden Service:
> 
> ```
> /home/admin/config.scripts/internet.hiddenservice.sh bitcoin8332 8332 8332
> ```
> 
> restart Tor or the whole node to rebuild the Tor circuits.

Perfect! thanks

closed 2021-05-04 by xpsx