raspiblitz/raspiblitz #2271: Dynu - Exception on Subscription

exfrog, 2021-05-04:
trying to setup https lets encrypt cert, thru the ssh menu 'subscription' options
create dynu.com account and setup ddns service
select dynu.com
enter dynu.com domain
enter 0auth2 client id
enter 0auth2 secret
select DYNDNS - HTTPS for DYNU DynamicIP DNS
get screen:
title: exception on subscription
Unknown Error happened - please report to developers:
local variable 'e' referenced before assignment

prior to getting this screen, terminal said something about:
# checking if Tor proxy for CURL is working ...
err='curl tor proxy not working'

additional info: never done a cert for this pi.  also just upgraded to 1.7.0

nicklyk, 2021-05-04:
I'm having a similar problem with my btcpay server.

Updated to 1.7.0
The link between my IP2TOR subscr. and DYNU didn't seem to work
Canceled all subscriptions
Created new IP2TOR Bridge for BtcPay
Tried to create LetsEncrypt for IP2TOR
After entering OAuth2 creds. i get the following response

`checking if Tor proxy for CURL is working ...`
`err='curl tor proxy not working'`
.
.
.
`('failed oAuth Service', 'https://api.dynu.com/v2/dns ErrorCode:401')`

exfrog, 2021-05-05:
Hrm, looks like whatever happened here has borked a lot of other features.
Not working:
sphinx, rtl, mempool, thunderhub (branded 404 page)
Working:
explorer, lnbits, btcpay, spectre

Heres my debug log https://termbin.com/b3o2

im not even sure where to begin trying to fix this. any recommendations?

exfrog, 2021-05-05:
Ok, continuing to troubleshoot, 
I tried a full reboot, but nothing changed.
I reflashed a new ssd and let the blitz rebuild itself.
now, all the above web interfaces are working, EXCEPT  RTL.
sphinx still gives me the error from the pi menu, but going to the url, i DO get INDEX response (which i didnt previously)
but the displayed qr code does not work to attach (freshly downloaded) clients to sphinx (ill troubleshoot with them once i confirm server is setup correct).  

Thunderhub error was my fault, wrong url. works now.

heres my full debug log  https://termbin.com/mtdc
my RTL log shows up empty.

my next instinct is to turn RTL off then back on again, but I wanted to check if anyone has any ideas first.

As far as I can tell, everything worked till i tried to do the cert and dyndns config.  But I tried the cert config shortly after upgrading to 1.7, so im not 100% sure which caused what.

any help would be appreciated. thanks!

exfrog, 2021-05-05:
partial solution found:

while looking thru other issues for anything that looked familiar, I came across this thread, and realized id seen these npm errors multiple times, (i believe since updating to 1.7, but not sure during what steps at this point...)
https://github.com/rootzoll/raspiblitz/issues/2124

following those instructions, i tried updating rtl, no success, 
then i toggled rtl off from the menu
and toggled it back on, got an ERROR telling me to manually run bonus.trl.sh on 
when i ran that command, it said rtl user still exists
deleted rtl user
ran script
rtl installed
works correctly.

ill leave this open while i investigate dynu. its possible i saw the npm error during that process too, and maybe this fixed it?

exfrog, 2021-05-05:
same err='curl tor proxy not working'

so ignore all the bs inbetween ;)

rootzoll, 2021-05-16:
This could be related to this patch here: https://github.com/rootzoll/raspiblitz/issues/2247#issuecomment-840688613

Can you try if this fixes it?

exfrog, 2021-05-18:
@rootzoll 
I ran the patch from the menu, rebooted, checked subscriptions, none active, and ran letsencrypt dynu setup for dyndns
same error
local variable 'e' referenced before assignment

rootzoll, 2021-05-18:
so it still says ` err='curl tor proxy not working'`?

exfrog, 2021-05-18:
nope, have not seen that message again since running the patch.

so, the first time, the term window sat at 'restarting ngnix' for a bit, and i didnt see if there were more messages before it switched to the blue screen with the local variable 'e' message.

after getting the error, i rebooted, checked subscriptions again (all off) and re-tried
terminal window flashed briefly (no messages afaik) then variable e error showed directly.

checked subscriptions (all off), rebooted, tried again,
this time i saw term window say 'lets encrypt has already been installed', then it flashed to the vairable 'e' error screen again.

exfrog, 2021-05-18:
Also, possibly tangently related, sphinx hasnt worked since 1.7 upgrade and letsencrypt attempt.
I get this message:
 Your SPHINX SERVER may have problems (retry if just restarted).      │
    │                                                                      │
    │ Check if locally responding: http://10.1.1.217:3300/app              │
    │ (You should see 'INDEX' in your browser)                             │
    │                                                                      │
    │ Check if service is reachable over Tor:                              │
    │ hj7a3nfxstgh4fwpjp7wujwzoalt7rikikarzasc3wp2uiablo4ipvqd.onion/app   │
    │                                                                      │
    │ If you use IP2TOR may cancel & renew subscription & LetsEncrypt.     │
    │ Also check logs with 'debug' on terminal.        

I get INDEX response over clearnet and tor.

also, fwiw, i get a tor exit node page when i go to my dynu domain, but need to investigate my port forwarding (it was working pre-letsencrypt attempt, but port may have changed?)

nicklyk, 2021-05-18:
@rootzoll I tried the fix you suggested. This time it didn't give me the aforementioned error `err='curl tor proxy not working'`.

On the other hand, it did throw some other errors resulting in this `('failed oAuth Service', 'https://api.dynu.com/v2/dns ErrorCode:401')`

Full log below:

```
no crontab for admin~
*** INSTALLING Let's Encrypt Client 'acme.sh' ***
[Tue May 18 20:04:19 BST 2021] Installing to /home/admin/.acme.sh
[Tue May 18 20:04:19 BST 2021] Installed to /home/admin/.acme.sh/acme.sh
[Tue May 18 20:04:19 BST 2021] Installing cron job
no crontab for admin
no crontab for admin
[Tue May 18 20:04:19 BST 2021] Good, bash is found, so change the shebang to use bash as preferred.
[Tue May 18 20:04:21 BST 2021] OK

# default IP certs
# default TOR certs
# certsDirectories(###########.ddnsgeek.com_ecc)
# i(###########.ddnsgeek.com_ecc)
# FQDN(##########.ddnsgeek.com)
list indices must be integers or slices, not str
# details(error='list indices must be integers or slices, not str')
parse error: Invalid numeric literal at line 1, column 12
# FAIL target '' not supported yet'
# restarting nginx
# dnsservice=dynu --> update ##########.freeddns.org
# dynu update IP API call
# domain(#########.freeddns.org)
# token(#########-####-####-####-##########:#################################)
# ip(##.###.##.###)
Splitting oAuth user & pass:
#################################
###############################
# calling URL: https://api.dynu.com/v2/oauth2/token
# headers: {'accept': 'application/json'}
# response-code: 200
b'{"access_token":"#####################################################||","token_type":"bearer","expires_in":28800,"roles":[]}'
# apitoken(##############################################################||)
# cleaning API token:
# apitoken(######################################################3)
# API CALL --> Getting ID for Domain (list all domains and search thru)
# calling URL: https://api.dynu.com/v2/dns
# headers: {'accept': 'application/json', 'Authorization': 'Bearer ##################################################'}
('failed oAuth Service', 'https://api.dynu.com/v2/dns ErrorCode:401')
```

# `#` <--- wherever needed for privacy reasons

exfrog, 2021-05-18:
ok, donno if this info helps, but when i try the letsencrypt dynu setup (and get the error above), it breaks lnd > 'unable to set up wallet password listeners: shutting down'
(found in debug log)

exfrog, 2021-06-13:
Alright, coming back to this:

menu says all subscriptions are off
when i try to install the lets encrypt, ddns, dynu.com, i enter all the credentials, then get this message:
# *** Let's Encrypt Client 'acme.sh' appears to be installed already ***

then, after quite some waiting:
Exception on Subscription
Unknown Error happened - please report to developers:
local variable 'e' referenced before assignment

attempting this breaks lnd, but a restart usually brings it back.

tangentially, this has borked my sphinx relay, which is what keeps bring me back to the issue.

any help would be appreciated, im still learning how to get around the blitz backend

thanks!

nicklyk, 2021-06-13:
Running the Update > Patch did it for me. I first unsubscribed from all of my subscriptions, updated to Bitcoin Core 21.1 (alhtough this might be unnecessary), ran the patch and then resubsribed to IP2TOR and added the LetsEncrypt to it.

When patching i saw that there were some updates being done to the scripts responsible for this issue of mine.

Personally, this issue is closed. :)

exfrog, 2021-06-13:
thanks. trying now.
this is probably a stupid question, but if im running tor, do i need the IP2TOR subscription before the letsencrypt cert will work?

we-do-it-lu, 2021-07-06:
Hello I am having the same issue here (LetzEncrypt HTTPS Domain with DYNU.
Followed the steps in the Menus to configure DYNU but I get "`local variable 'e' referenced before assignment`" on the last screen.
Then I found this issue #2271, see that someone solved it patching.
So I patched and rebooted.
Not resolved. Any idea?

exfrog, 2021-07-06:
I am still having this issue, too.
Ran updates, patches, enabled and disabled everything, reflashed ssd
multiple times.

Trying the let's encrypt flow has broken my sphinx relay completely, and my
lnd requires a restart or two to come back online, so in respect to my
lightning counterparties boz scores, Ive stopped trying till I get more
clarity.

Not afraid to muck about in the cli, but also not experienced enough to
know where/what to poke at.

Thanks in advance for any suggestions

HardCorePawn, 2021-07-09:
Having the same issue trying to get Let's Encrypt with DYNU ddns subscription working.

After attempting the "Patch" fix... I at least got to something involving installing acme.sh... but it failed, LND was stuck and I had to reboot to try and get things working again.

Now, if I try to add the subscription again, I am stuck at:
> *** Let's Encrypt Client 'acme.sh' appears to be installed already ***

Then a looooooooooooong pause and then the following dialog is displayed:
> Exception on Subscription
> Unknown Error happened - please report to developers:
> local variable 'e' referenced before assignment

Any ideas?

rootzoll, 2021-08-17:
The Dynu issue is still on my fix list for the v1.7.1 and willbe addressed after the RC1

rootzoll, 2021-09-22:
I was able to fix the API calls with dynu to update the IP but the ACME script fails on processing. 
Will deactivate `dynu` for v1.7.1 and see if we can maye switch for v1.8.x to an ACME implementation go - see:
https://go-acme.github.io/lego/