raspiblitz/raspiblitz #2272: Telegraf show wrong CPU temperature

AllYouCanFuck, 2021-05-04:
Telegraf shows the wrong CPU temperature in Grafana.

CPU:
[image]
GPU:
[image]

PatrickScheich, 2021-05-05:
You are totally right, 

- `cpu_temperature` returns milli celcius as an integer.
- `gpu_temperature` returns celcius as a float

to get the proper CPU temperature in celcius you can use this...
[image]

AllYouCanFuck, 2021-05-05:
Works! Thanks!

closed 2021-05-05 by AllYouCanFuck

rootzoll, 2021-05-16:
@PatrickScheich happy to merge a PR on this for the v1.7.1