raspiblitz/raspiblitz #2273: SSH over Tor (torify ssh?)

peterflock, 2021-05-04:
I like to connect to my Raspiblitz over Tor when I am not at the place (on LAN).

I have found this [guide on JoininBox](https://github.com/openoms/joininbox/blob/master/FAQ.md#ssh-through-tor-from-linux).

My suggestion is:
- Link this in Raspiblitz Readme/Doc/FAQ so everyone is aware of this option.
- Maybe add it to menu in Raspi (I think this should be quite a common use-case - the need to manage note while on the road)
- Maybe run ssh as tor-service by default? Is there any security concern?

nyxnor, 2021-05-05:
>Link this in Raspiblitz Readme/Doc/FAQ so everyone is aware of this option.

Yes

>Maybe add it to menu in Raspi (I think this should be quite a common use-case - the need to manage note while on the road)

Just FAQ, cause it is just one simple command

>Maybe run ssh as tor-service by default? Is there any security concern?

No, lots of concerns. This is for experienced users, cause if someone share this address (and believe me, they share by mistake some of their hidden service addresses on telegram every week), they need to understand the consequences. There is still fail2ban, but no uncessary open ports to the internet.

nyxnor, 2021-05-16:
Also, optionally (at least for now), you can add 1 extra security layer with client auth.
https://gist.github.com/mtigas/9c2386adf65345be34045dace134140b

I will make a pull request later this month hopefully introducing this. to `internet.tor.sh`, so even if the address got leaked, the intruder can't complete the connection, meaning he does not even have the option to try your user password. This is kinda a 2FA, but as it is on the same device you are using, not the same level of protection when using TOTP. You can try to implement a ssh 2FA https://github.com/rootzoll/raspiblitz/issues/577

nyxnor, 2021-07-22:
please close this

closed 2021-08-19 by rootzoll