raspiblitz/raspiblitz #2277: Electrum Server failed to start

ronledfordjr, 2021-05-05:
RPi 4B 8GB
Raspiblitz 1.7
SSD 1TB

https://termbin.com/im2v5

I am trying to install again.

ronledfordjr, 2021-05-05:
2nd install
[image]
 failed

openoms, 2021-05-05:
We will need to use a more recent release of Rust: https://github.com/romanz/electrs/blob/master/doc/usage.md#build-dependencies

Can try to install the latest Rust with:
`sudo apt install cargo`

and uncomment (put a # in front of) the line:
https://github.com/rootzoll/raspiblitz/blob/be19c565cb5fb570abcd7cf6ad83cc5e4b840520/home.admin/config.scripts/bonus.electrs.sh#L266
in :
`nano config.scripts/bonus.electrs.sh`

Now installing electrs again should work.

ronledfordjr, 2021-05-05:
Thank you. Should I update the other two nodes?

openoms, 2021-05-05:
Sorry, what other two nodes?

electrs on 1.7 works, I couldn't reproduce your issue, but if you have this problem now there is a solution.

ronledfordjr, 2021-05-05:
I have three raspiblitz 1.7 nodes.

ronledfordjr, 2021-05-09:
Thank you for the help

closed 2021-05-09 by ronledfordjr