raspiblitz/raspiblitz #2280: RaspiBlitz v1.7.0-> .onion-address missing

silvester235, 2021-05-05:
I recently updated to RaspiBlitz v1.7.0.

Before v1.7.0 the "Raspiblitz Status Screen" showed the .onion-address
However under v1.7.0 this not displayed anymore. Where can I find the .onion-address?
[image]

openoms, 2021-05-05:
`bitcoin-cli getnetworkinfo`

silvester235, 2021-05-05:
@openoms thanks!

closed 2021-05-05 by silvester235

rootzoll, 2021-05-05:
Just for background: Because the short Torv2 addressed are deprecated and now the longer Torv3 addresses are used by bitcoin core the onion-adress got too long to display on the LCD and I replaced it on the display with the block number information. But of course as openoms pointed out you can find that information still by calling bitcoin-cli directly.