raspiblitz/raspiblitz #2282: Suggestion re: file rescue output

originaljeremy, 2021-05-06:
Right now, when a user performs a file rescue operation, the output is:

****************************
* DOWNLOAD THE RESCUE FILE *
****************************

ON YOUR MAC & LINUX LAPTOP - RUN IN NEW TERMINAL:
scp -r 'admin@192.168.1.X:/home/admin/lnd-rescue-*.tar.gz' ./
ON WINDOWS USE:
scp -r admin@192.168.1.X:/home/admin/lnd-rescue-*.tar.gz ./

Use password A to authenticate file transfer.
Check for correct file size after transfer: 123456789 byte

BEWARE: Your Lightning node is now stopped. It's safe to backup the data and
restore it on a fresh RaspiBlitz. But once this Lightning node gets started
again or rebooted its not adviced to restore the backup file anymore.

Press ENTER when your backup download is done to shutdown.

LCD turns white when shutdown complete.
Then wait 5 seconds and disconnect power.
-----------------------------------------------

I recommend a slight change to the following:

****************************
* DOWNLOAD THE RESCUE FILE *
****************************

ON YOUR MAC & LINUX LAPTOP - RUN IN NEW TERMINAL:
scp -r 'admin@192.168.1.6:/home/admin/lnd-rescue-*.tar.gz' ./
ON WINDOWS USE:
scp -r admin@192.168.1.6:/home/admin/lnd-rescue-*.tar.gz ./

Use password A to authenticate file transfer.

Type the following command into the new terminal:
stat lnd-rescue-*.tar.gz

Check for correct file size after transfer: 77515365 byte

BEWARE: Your Lightning node is now stopped. It's safe to backup the data and
restore it on a fresh RaspiBlitz. But once this Lightning node gets started
again or rebooted its not adviced to restore the backup file anymore.

Press ENTER when your backup download is done to shutdown.

LCD turns white when shutdown complete.
Then wait 5 seconds and disconnect power.
-----------------------------------------------

The rationale being that a RaspiBlitz user may not immediately remember an appropriate command to display file size and most prompts for scp will round the size to MB, which can be confusing if you expect the output to match exactly as you give it. Also, it would be a good time to change 'adviced' to 'advised' and 'complete' to 'completes' to fix those small typos.

nyxnor, 2021-05-06:
#### About typos:
* Step 1 FORK
* Step 2 https://github.com/rootzoll/raspiblitz/pulls > New Pull Request

#### About file size: https://github.com/rootzoll/raspiblitz/issues/1974#issuecomment-769442762
>The lnd rescue file already contains the md5 checksum as part of the filename:
https://github.com/rootzoll/raspiblitz/blob/82e0d6c3714ce1b2878780c4bdef72a6417f71c7/home.admin/config.scripts/lnd.rescue.sh#L44

originaljeremy, 2021-05-07:
typos were fixed
the file size comment you link doesn't speak to the concern.
i'm not suggesting there isn't a way to verify file size and am aware the filename contains the checksum
the user is unlikely to know this or act upon that knowledge if they have it
simply instructing them to use stat will output bytesize and give the user an unambiguous confirmation that the file is the correct size
I have submitted a pr to that effect

closed 2021-05-07 by originaljeremy