raspiblitz/raspiblitz #2285: What are the requirements for LiteCoin?

jgabriel98, 2021-05-07:
I've just discovered thuis topic, and saw somewhere that you need a 1TB SSD for bitcoin LN, since the blockchain is big and will grow.

But what are the HD/SSD spaces needed for litecoin LN?
and a SSD is that essential, or can i use a hard drive?, or a usb stick? or even the SD card itself?

rootzoll, 2021-05-08:
Using the sd card for blockchain storage in general is not recommended and I dont think that a raspberryPi zero will give you enough performance ... RaspberryPi 3 is really the lower limit and RP4 recommended (you want USB3).

A 500GB SSD should be fine, HDD will work too but is much slower. USB sticks are not recommended for safe data storage.

cstenglein, 2023-06-07:
@jgabriel98 Were your questions answered? 

Closing this for now, if you got more questions, please re-open the issue / make a new one. :)

closed 2023-06-07 by cstenglein