raspiblitz/raspiblitz #2288: LCD Screen - 2nd line missing node address, terminal session

ronledfordjr, 2021-05-09:
I have one node missing part of its node address on terminal but shows on LCD screen. Hopefully not a big deal.

[image]

[image]

cstenglein, 2021-05-09:
I usually use ssh with my Linux PC and can't reproduce the issue there.

I see that you are using PuTTY(?), but I can't reproduce it there as well with a windows machine with PuTTY 0.75 (got an issue there with q's & x's showing in the main menu but other than that shows the address fine).

What OS and Software are you using to access the session?

cstenglein, 2021-05-10:
No worries @ronledfordjr , happy to help :)