raspiblitz/raspiblitz #2290: Paralell testnet and signet services

openoms, 2021-05-10:
Using the same binary but with different  systemd services it is possible to have mainnet, testnet and signet run parallel.

Similarly other services can be connecred and running alongside the instances connected to mainnet.

Proposing to make it for bitcoin core first by having entries in the SERVICES menu for signet and testnet.

Activating these would not require to shut down the mainnet instance.

Once TESTNET/SIGNET is active a new main menu entry could be used to set up other individual services similarly in a paralell, but on different ports.

Will first implement only for the minimum like lnd, clightning and RTL and can add further services later.

openoms, 2021-05-11:
the RPi4 + SSD seems to take the three networks well. 
Testnet IBD is now done without hiccups ~ 10 hours.

The biggest advantage to this setup is that all can remain in sync and changing the defaults can be quick.
Using different ports for the services on different networks means the already running services can carry on and the connections are not disturbed.
The existing Hidden Services and connected wallets can continue to function and there is no confusion in which network they are connected to.

### Ports
aiming to follow the logic of the Bitcoin ports.
Services which use a 4 digit port number:
- Testnet services will have a leading 1 added
- Signet services will have a  a leading 3 added

Services with 5 digit port numbers:
- will aim to change the first digit to 1 or 3 accordingly
- if already starting with 1 or 3 a unique port will be chosen

### raspiblitz.conf
the entries:
`mainnet=on`
`testnet=on`
`signet=on`
can be added so they can be selectively switched on and off.

The defaults will be continues to be governed by:
`network=bitcoin`
`chain=main | test | sig`

openoms, 2021-06-27:
Testing on the latest state in https://github.com/rootzoll/raspiblitz/pull/2352: 
REPO: `openoms`
BRANCH: `cblitz-merge-dev`

1 bug after starting the parallel TESTNET from SERVICES:
```
sudo systemctl status lnd
● lnd.service - LND Lightning Daemon                                               PASSWORD   Change Passwords                                                                            │ │  
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)EPAIR     Repair Options                                                                              │ │  
   Active: activating (auto-restart) (Result: exit-code) since Sun 2021-06-27 17:58:17 BST; 35s ago/Prepare RaspiBlitz Update                                                             │ │  
  Process: 22714 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)                                                           │ │  
  Process: 22716 ExecStart=/usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --tor.socks=9070 --tor.control=9071 --listen=127.0.0.1:9735 ${lndExtraParameter} (code=exited, status=1/FAILURE)
 Main PID: 22716 (code=exited, status=1/FAILURE)──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘ │  

admin@raspberrypi:~ $ sudo -u bitcoin /usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --tor.socks=9070 --tor.control=9071 --listen=127.0.0.1:9735
Attempting automatic RPC configuration to bitcoind                             <Select>                              < Exit >                                                               │  
unable to load RPC credentials for bitcoind: unable to extract RPC credentials: unable to find zmqpubrawblock in config, cannot start w/o RPC connection────────
```

Solution:
`./config.scripts/lnd.chain.sh on mainnet` 
parses the min lnd.conf to add the correct zmqpubrawblock values.

rootzoll, 2021-10-11:
OK basic switching tested OK .. closing for final release.