raspiblitz/raspiblitz #2295: C-lightning on the RaspiBlitz

openoms, 2021-05-11:
Will share the development of the c-lightning implementation here.

The first functional install script is in:
https://github.com/openoms/raspiblitz/blob/cblitz/home.admin/config.scripts/cln.install.sh

The aim is to have c-lightning first as an optional service and later as a full featured alternative of LND.

The development is helped by being able to run signet and testnet parallel as in: https://github.com/rootzoll/raspiblitz/issues/2290

openoms, 2021-05-23:
Many new things are functional now:
- [ ] CLN menu for basic lightning actions like
- [ ] FUNDING
- [ ] PEERING
- [ ] CHANNEL
- [ ] PAY
- [ ] REQUEST

- [ ] Sparko plugin - providing a web interface and connection to Zeus / Spark Wallet
- [ ] C-Lightning-REST for RTL

- [ ] further improvements of running the chains and LND/CLN services parallel

openoms, 2021-06-01:
The status screen and parallel networks are functional.
If syncing with https://github.com/openoms/raspiblitz/tree/cblitz or https://github.com/rootzoll/raspiblitz/tree/pre-1.7.1 make the new scripts executable with:
`chmod +x *.sh`
and a restart is necessary.

For C-lightning the keys will be generated from 24 words which will be created by the BIP39 reference implementation: https://github.com/trezor/python-mnemonic#python-mnemonic. This way the on-chain wallet backup will be similar to LND and other wallets.

Automating the seed creation, encryption, decryption and auto-unlock is depending on the solution of https://github.com/ElementsProject/lightning/issues/4570.

openoms, 2021-06-02:
Next up:

- [ ] encrypt / decrypt / autounlock
- [ ] Summary plugin fro channel stats in menu
- [ ] Sparko hidden service
- [ ] Zeus pairing with QR to Sparko over Tor

openoms, 2021-06-28:
- [ ] A round of testing is due after some heavy refactoring.

Outstanding:
- [ ] Zeus pairing in menu
- [ ] encrypt / decrypt / autounlock the `hsm_secret` with the `hsmtool`
- [ ] Suez https://github.com/rootzoll/raspiblitz/issues/2366
- [ ] backup the `lightningd.sqlite3` to the SDcard periodically (at least on raspiblitz restarts) with `cln-plugin.bakcup.sh`

openoms, 2021-08-10:
https://github.com/rootzoll/raspiblitz/commit/52d71f74eedbeedc8fdd05af2a40bea55027e222 adds the CLBOSS plugin - The C-Lightning Node Manager
CLBOSS is an automated manager for C-Lightning forwarding nodes.
more info: ttps://github.com/ZmnSCPxj/clboss#operating

install with:
`config.scripts/cln-plugin.clboss.sh on`

openoms, 2021-08-10:
[image]
[image]

rootzoll, 2021-08-16:
@openoms do you think a QR code connection with Zeus for c-lightning is possible within the next 2 weeks or should we target for the v1.7.1 release a workaround - like a short text describing the user how to connect manually?

openoms, 2021-08-16:
@rootzoll I think the best we can do is to coordinate with the Zeus team in https://github.com/ZeusLN/zeus/issues/515 to ask them to implement a QRcode parsing the way it will be implemented in the RaspiBlitz.

Spark-wallet already does use a QRcode format, but it needs a CA signed certificate (or importing the certificate in the phone which is far from easily atm).
Should try if it can be used with `LetsEncrypt HTTPS Domain` subscription.

rootzoll, 2021-08-17:
@openoms On a test build of the modified sd build script with c-lightning I ran into this that made the build stop:
```
  inflating: clightning-v0.10.1/wire/wire_sync.h
- Configuring EXPERIMENTAL_FEATURES enabled
Compiling ccan/tools/configurator/configurator.../usr/bin/ld: cannot open output file ccan/tools/configurator/configurator: Permission denied
collect2: error: ld returned 1 exit status
- Building from source C-lightning
File config.vars not found: you must run ./configure before running make.
make: *** No rule to make target 'config.vars', needed by 'show-flags'.  Stop.
- Install to /usr/local/bin/
File config.vars not found: you must run ./configure before running make.
mkdir -p /bin
mkdir -p /libexec/c-lightning
mkdir -p /libexec/c-lightning/plugins
mkdir -p /share/man/man1
mkdir -p /share/man/man5
mkdir -p /share/man/man7
mkdir -p /share/man/man8
mkdir -p /share/doc/c-lightning
make: *** No rule to make target 'config.vars', needed by 'ccan/config.h'.  Stop.
```

rootzoll, 2021-08-17:
It runs when using just `sudo` @openoms is there a specific reason why building it with `sudo -u admin`

openoms, 2021-08-17:
@rootzoll wanted to avoid to run the compilation with root, but does not make much difference really.
I have tested the script from the admin user, not pi, which could be the difference.

openoms, 2021-08-17:
It does not need root according the docs either:
[image]
https://lightning.readthedocs.io/INSTALL.html#to-build-on-ubuntu

openoms, 2021-08-18:
> @openoms do you think a QR code connection with Zeus for c-lightning is possible within the next 2 weeks or should we target for the v1.7.1 release a workaround - like a short text describing the user how to connect manually?

I think since the QR format is still in flux we should show the access-key or macaroon in itself in a QR and put the explanation to paste bit manually to Zeus.

Can have two steps switching the QRcode on the LCD after enter:
first would just show the access-key or macaroon
second the onion address

so they can be scanned individually and pasted in the app manually.

rootzoll, 2021-08-19:
@openoms sounds good as an temp workaround. Will you work on that?

openoms, 2021-08-19:
> @openoms sounds good as an temp workaround. Will you work on that?

sure will sort it

rootzoll, 2021-08-24:
This will be contained in v1.7.1rc1 for further intense testing.

openoms, 2021-08-31:
Added a CLNREPAIR menu to cover the functions of `cln.hsmtool.sh` in the dialogs
[image]
[image]

The `hsm_secret password` is equivalent to the `Password_C` for LND.
By default the `hsm_secret` is stored in cleartext on the disk.
If encrypted the `hsm_secret password` is kept in memory between restarts. 
The `AUTOUNLOCK` means that the `hsm_secret password` is stored on the disk in cleartext.

Creating a new C-ligtning wallet will now create a `seedwords.info` beneath the `hsm_secret` with the contents:
```
# This file was placed by cln.hsmtool.sh .
# Contains the seed words from which the hsm_secret in the same directory was generated
seedwords='${seedwords}'
seedwords6x4='${seedwords6x4}'
# Will be removed safely when the hsm_secret is encrypted.
```
```
# This file is placed by cln.hsmtool.sh .
# The seed words from which the hsm_secret in the same directory was generated
# were deleted when the hsm_secret was encrypted.
# The words cannot be generated from the hsm_secret (one way function).
# If you don't have the words the hsm_secret can be still backed up in hex:
# https://lightning.readthedocs.io/BACKUP.html#hsm-secret 
```

openoms, 2021-08-31:
Some simpler tasks to do next:
- [ ] CLBOSS - Automatic channel management for C-lightning
- [ ] ADDPLUGIN  - Add a custom binary as a C-ligthning plugin
- [ ] LISTCLOSED - List closed channels with details
- [ ] SPARK - Spark Wallet to manage the C-Lightning wallet (possible replacement for Sparko)

rootzoll, 2021-09-09:
OK added "display-seed" option to `cln.install.sh`that now will be called if user activated the secondary/parallel cln wallet.

openoms, 2021-09-20:
Future integrations:

- [ ] LNBits backed by CLN

rootzoll, 2021-09-20:
@openoms for the CLN LNbits best would be to open a ticket on v1.8.0 milestone.

rootzoll, 2021-10-11:
OK .. so lets render this as an edge case for now and see what reports come in on this after final release. If there is something we didnt see we can always do a quick 1.7.2 before 1.8.0 - closing this issue.

closed 2021-10-11 by rootzoll