raspiblitz/raspiblitz #2296: Recommended SanDisk SSD PLUS issue ...

sirenak2, 2021-05-12:
Been setting up a new blitz using the recommended items from amazon. The ssd, drive bay and micro sd card. After having to redo the set up I discovered it was the SSD. Seems like it's dead on arrival. Has anyone else had issues with this drive? I read some amazon reviews and saw quite a few saying the drive dies after a few months or just doesn't work. I'll have to return it, not sure I trust trying it again...any recommendations?

openoms, 2021-05-12:
I found this drive quite reliable.

Try to format on the dekstop first to see if it works and start the raspiblitz setup again.

If DOA Amazon takes it back for 14 without questions asked.

sirenak2, 2021-05-12:
I tried it on the pi,  laptop and desktop also connected directly with data, nothing.  I'm getting it replaced,  fingers crossed..

closed 2021-05-12 by openoms