raspiblitz/raspiblitz #2297: 1.7 update, wifi broken

woeisme, 2021-05-13:
Just updated to 1.7 and seem to have lost the ability to run the node via Wifi. 

Copied the wpa_supplicant.conf file from /etc/wpa_supplicant/wpa_supplicant.conf  > /boot as old raspibolt instructions say but nothing. Never needed to do this in 1.6 anyway. 

Possibly need to change something in /etc/default/network perhaps?

woeisme, 2021-05-14:
Adding a thread which might be related [here](https://raspberrypi.stackexchange.com/questions/112062/raspbian-buster-lite-couldnt-communicate-with-wpa-supplicant)

woeisme, 2021-05-14:
annnd it was something stupid ofc!

after update country code in wpa_supplicants reverted back to (not actually AQ, just using example here)
`country=[AQ]`
when it should read
`country=AQ`

woeisme, 2022-08-09:
wifi settings (specifically wpa_supplicant) is still not surviving upgrade process between versions (recently upgraded to 1.8.0)

rootzoll, 2022-08-10:
Marked to retested for v1.8.1

rootzoll, 2022-12-11:
OK I retested for upcoming version 1.9.0 - it kept the Wifi config wihile recovering/updating. 
I keep this issue open so that you can report on your experiences once the Release Cadidate version is out.
Please check before your update if the following file exists on your HDD/SSD:
```
/mnt/hdd/app-data/wpa_supplicant.conf
```
Basically thats the backup of your Wifi config on your HDD/SSD that will be reinstalled during an update.

rootzoll, 2025-02-26:
USER REPORT: heya, upgraded to 1.11.3 (from 1.10) tonight lost wifi settings again like my old ticket

- [ ] retest for v1.12.0

woeisme, 2025-03-29:
> USER REPORT: heya, upgraded to 1.11.3 (from 1.10) tonight lost wifi settings again like my old ticket
> 
>     * [ ]  retest for v1.12.0

slight update here since I regained GH access, it was related to updated RPI country specific wifi rx/tx errors that were solved by running raspi-config > Localization Options > WLAN Country Set.

please note that wireless settings however still did **not** get copied during migration.

rootzoll, 2025-05-17:
@woeisme can you discribe again how wifi gets lost on migration - because in general a backup is stored in /mnt/hdd/app-data/wifi an that directory should be part of every migration file and used on new setup.

Also how was the wifi before set? Thru internet.wifi.sh script or different?

Also how important is the Wifi country setting - does a wrong (or default US country) give you errors?