raspiblitz/raspiblitz #2304: Adopt infoscreen text size to touchscreen resolution/size

subyy222, 2021-05-15:
Current text size does not consider the touchscreen size. Works fine on 3.5" touchscreen, but not on 5" or 7".
Would like to see a option to scale with a factor (e.g. x2) or configuration file for common screen sizes.

rootzoll, 2021-05-18:
There are bigger changes coming for the v1.8 to the touchscreen/LCD that should make it easier to scale to different screen sizes. For details maybe @fusion44 likes to comment?

fusion44, 2021-05-19:
My plan is to write an app for all kinds of devices and create it in such a way that it is flexible enough to adapt to different screen sizes and forms. This means that I don't just want to increase the font size but also adapt the information displayed to the screen size currently running. My first priority is the 3.5" screen as it is currently the default recommended screen for the Blitz. This should be ready for 1.8 everything else is a more long term activity :-)