raspiblitz/raspiblitz #2306: SSH Tor serivce is not prsisted after update

peterflock, 2021-05-16:
Related to https://github.com/rootzoll/raspiblitz/issues/2273

I added SSH Tor service for allow to connect to the node following this process https://github.com/openoms/joininbox/blob/master/FAQ.md#ssh-through-tor-from-linux

After update, the ssh tor service is no longer accessible.

I like to suggest to persist this, because it may leave someone (like me now) lock out of the node when on the road.

nyxnor, 2021-05-16:
Do you mean persist the hostname after update?

>No longer accessible.

See Tor logs: `sudo journalctl -eu tor@default` to make sure it is functional.

peterflock, 2021-05-17:
I had to do `./config.scripts/internet.hiddenservice.sh ssh 22 22` again.

Onion address was the same (hostname was persisted).

nyxnor, 2021-05-17:
If the hostname was the same, it did not create a new address, it used the same priv key. Probably was a Tor issue.

openoms, 2021-05-18:
the torrc is on the sdcard so settings get lost on OS updates.
Not a bad thing tbh since during the recovery there is a temporary revert to the default password. Better to not have remote access then.

nyxnor, 2021-05-18:
@openoms I see your point of the password being reverted to default can be a security issue when open ports to the internet. If using client auth with Tor, there is no problem with this, as the authentication comes before asking the password, so this is pretty secure, but not implemented on the code yet to advise to be done for now. So for the future, just want to keep this in mind.

peterflock, 2021-05-18:
Good point with security after pw reset. However, can there be at least some warning after the first login (post update, post password change)? Something saying: "You had Tor SSH, but it was disabled during update. Please run `<this command>` to enable it again."

Because what happend to me, was that I did update, was happy that everything works so smoothly. Then started doing some stuff and then left home with plan to finish it later remotely. I didn't realize this and had to drive back home from quite far away.

nyxnor, 2021-05-25:
@peterflock By update away from the node, what do you mean? `Patch` command?

nyxnor, 2021-05-26:
Actually, I think the service was still available, what happened is that Update > Release turns of the pi https://github.com/rootzoll/raspiblitz/blob/f571dca4857b800740f8c8e5fc980f32a0fcdb92/home.admin/99updateMenu.sh#L76
So you couldnt ssh because the pi was off, not the ssh Tor service.

But okay, when flashing the sd card when doing update at home, maybe insert ssh here https://github.com/rootzoll/raspiblitz/blob/f571dca4857b800740f8c8e5fc980f32a0fcdb92/home.admin/config.scripts/internet.tor.sh#L380
For now, you will have to use `custom-installs.sh`.

peterflock, 2021-05-28:
After update I have SSHed on local network. I have also installed and tested new LIT. Then I hit the road and realized I cannot ssh over Tor.

nyxnor, 2021-05-28:
>After update I have SSHed on local network.

So you turned on the pi manually. If you were traveling, this would not be an option. About letting ssh with tor enabled after update, more below:
Oh, I see, the torrc file is replaced, /mnt/hdd/tor/ssh is still there, but not configured in the torrc. Use `custom-installs.sh` for now or fork Blitz and change your `internet.tor.sh` as mentioned above.
Openoms has some concerns with letting open ports and in this case, the most significant (ssh), so I will wait.

nyxnor, 2021-12-01:
I plan to fix this on 
- #2711 

the user hiddenservices will not be overwritten, need to be created in a separate way

rootzoll, 2021-12-13:
Use this as test case to test if #2711 keeps hidden services over updates/restores.

rootzoll, 2022-02-17:
The script to create the hidden Tor service is now called: `/home/admin/config.scripts/tor.onion-service.sh ssh 22 22`
The keys & configuration should from v1.7.2 on be persistent between updates.

Closing this issue for release.

closed 2022-02-17 by rootzoll