raspiblitz/raspiblitz #2308: invalid handshake version: 22, only 0 is valid

The34Goon, 2021-05-16:
Seeing many entries like this in the LND log.

**[ERR] BTCN: Can't accept connection: unable to accept connection from 127.0.0.1:60368: act one: invalid handshake version: 22, only 0 is valid**

like every 5seconds on average.

Pretty sure its related to some of my channel partners disconnecting from me and remain unreachable. *shrug

Tried:

- Recreating certs
- Installing Tor from source
- IP2Tor for clearnet IP
- Flashing a new SD

but nada.

Debug: https://termbin.com/ysr1

How can I fix that?

rootzoll, 2021-05-18:
@openoms any idea? Or maybe something to ask at the LND dev slack? https://dev.lightning.community/

openoms, 2021-05-18:
Must be some erroneous setting in the configuration: https://github.com/lightningnetwork/lnd/issues/1173

I would check the 
`/mnt/hdd/lnd/lnd.conf`
first.

See the raspiblitz defaults: https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/assets/lnd.bitcoin.conf

The34Goon, 2021-05-25:
Unfortunately nothing has changed with the replacement of the conf. I'll close this anyway and see what can be found over at LND

closed 2021-05-25 by The34Goon