raspiblitz/raspiblitz #2310: SatSale - BTCPyment

nyxnor, 2021-05-17:
https://github.com/nickfarrow/SatSale

SatSale is a simple, easily deployable, lightweight Bitcoin payment processor that connects to your own Bitcoin node or Lightning network node. Lightweight Bitcoin payment processor written in easily deployable Python. Add Bitcoin payments to your Woocommerce webstore, or create donation buttons that are directly connected to your own Bitcoin node! 

SatSale currently serves as
* Donation button for your website that you can easily embed/link to anywhere.
* A Bitcoin payment gateway, including a Woocommerce plugin that easily turns ANY Wordpress site into a Bitcoin accepting store.

nyxnor, 2021-06-26:
Working on this today.... it would still require manual steps to configure in a self hosted domain not hosted in the blitz, as you would need to insert a line in the html file.

nickfarrow, 2021-06-28:
Awesome, let me know if you need a hand with anything. Or if there are any changes (eg config) that I can include to make things more compatible on the raspiblitz end.

I started making a docker branch for another node OS which mostly works. One feature to call out is that on chain payments are still made through lnd in the node interface. So in this branch I made onchain payments go through lnd with `newaddress p2wkh` rather than bitcoind RPC. (untested)
https://github.com/nickfarrow/SatSale/commit/5b42c5519b148c116ca505ded92fe574adc52ce5

nyxnor, 2021-06-28:
Did connect to the RPC credentials
[image]

But I did something wrong (using the templates/index.html)
[image]

nickfarrow, 2021-06-29:
Strange, I have had some installation issues with eventlet in the past. Perhaps try run without that flag?
Alternatively, I think the current version of SatSale uses an older version of eventlet (requirements.txt), perhaps the latest version works now.

BitcoinQnA, 2022-03-07:
Would love to see this implemented into Raspiblitz

nyxnor, 2022-03-07:
This is something I should look again and take time to code, but anyone else is free to do it also.

nickfarrow, 2022-03-09:
Thank you both. I'm keen to get this running also, regrettably I have not yet tried raspiblitz. Keen to migrate an umbrel node I was using for development, I suppose ill also need to get a screen (unless hdmi mouse & keyboard works?).

From what I can see the main tasks will be:
* Create config script like [btc-rpc-explorer.sh](https://github.com/rootzoll/raspiblitz/blob/320015dc6086fb2bf238527d67581eaa574ac638/home.admin/config.scripts/bonus.btc-rpc-explorer.sh)
* Create nginx config

among some other scripts and docs.

@nyxnor  did you make any start on a config script? 
Any suggestions of an app that might have a config script similar to what we need?

rootzoll, 2022-03-09:
FYI: There is now a template of a config script you can use to add your own app - in the comments it basically walks you thru the setps: https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/config.scripts/bonus.template.sh

nickfarrow, 2022-03-09:
Perfect, that's exactly what i was looking for