raspiblitz/raspiblitz #2311: was HDMI info screen output dropped in 1.7??

DrBlackross, 2021-05-17:
just wondering, i can login and everything works ok, but the info screen doesn't pop up on the external monitor. 

i edited the config.txt
# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=1
hdmi_mode=3
gpu_mem=128

and ran

hdmi on

output is.....

"admin@raspberrypi:~ $ hdmi off
# SWITCHING VIDEO OUTPUT TO --> HDMI
# blitz.display.sh set-display hdmi
# old(hdmi)
# new(hdmi)
# nothing to uninstall - hdmi is the default/clean mode
# nothing to install - hdmi is the default/clean mode
file does exists

LCD turns white when shutdown complete.
It will then reboot again automatically."

still nothing on the external monitor

(btw the Nintendo case still works great lol)

DrBlackross, 2021-05-17:
(ok i can't rip oout the formatting above, lol)

rootzoll, 2021-05-18:
With v1.7.0 there are three command available to switch to different output modes:

- `hdmi` --> deactivats LCD and activates HDMI output
- `lcd` --> deactivates HDMI and activates LCD output
- `headless` --> deactivates LCD & HDMI (to run completly without any output device)

So @DrBlackross there is no need to edit the config.txt ... even doing so might mess up the switching between those modes. Maybe try `lcd` first and then again try to activate `hdmi` again. If thats not workling you might wanna make a fresh sd card and recover to have a clean output setup and try `hdmi` again. Just tested on my setup again and the v1.7.0 when ran the `hdmi` command was showing me the info screen after reboot.

DrBlackross, 2021-05-19:
explains why "hdmi on"  didnt work right lol

DrBlackross, 2021-05-19:
wait, still didn't work 

admin@raspberrypi:~ $ hdmi
# SWITCHING VIDEO OUTPUT TO --> HDMI
# blitz.display.sh set-display hdmi
# old(hdmi)
# new(hdmi)
# nothing to uninstall - hdmi is the default/clean mode
# nothing to install - hdmi is the default/clean mode
file does exists

LCD turns white when shutdown complete.
It will then reboot again automatically.
-----------------------------------------------
stop electrs - please wait ..
stop lnd - please wait ..
stop bitcoind (1) - please wait ..
Bitcoin Core stopping
stop bitcoind (2) - please wait ..

DrBlackross, 2021-05-19:
yea, all there is the boot process on the screen

[image]

but thunderhub works and the node does register online as being online with open channels

DrBlackross, 2021-05-20:
hmmm tried 
headless, lcd, hdmi.... all three on both hdmi ports (just to make sure)
still stops at that  screen shot above, with no info.

what file does the tweak, and what gets tweaked, so i can check the permissions

DrBlackross, 2021-05-20:
ahhhh.... no hdmi switch?

admin@raspberrypi:~ $  tail -n1000 -f raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.7.0
Thu 20 May 16:46:46 BST 2021
***********************************************
No HDMI switch found.
isMounted: 1
Checking for UASP deactivation ...
Check if HDD contains configuration ..
load and update publicIP
# OK internet detected ... continue
running LND users credentials update
###### SYNCING MACAROONS, RPC Password AND TLS Certificate ######
# make sure LND app-data directories exist

# copy macaroons to central app-data directory and ensure unix ownerships and permissions
# make sure admin has a symlink at ~/.lnd to /mnt/hdd/app-data/lnd/
# make sure LND conf is readable and symlinked
# make sure TLS certificate is readable and symlinked
Checking if additional backup device is configured .. (D54A-8248)
Yes - Mounting BackupDrive: D54A-8248
### blitz.backupdevice.sh ###
# BACKUP DEVICE MOUNT
# OK BackupDrive mounted to: /mnt/backup
isMounted=1
CLEANING TEMP DRIVE/FOLDER
OK: Temp cleaned
DONE BOOTSTRAP

DrBlackross, 2021-05-20:
GOT IT, had to uncomment
hdmi_safe=1
hdmi_drive=2

in the  /boot/config.txt
info screen is on the hdmi

closed 2021-05-21 by rootzoll