raspiblitz/raspiblitz #2312: lnd[1475]: unable to create chain control: unable to connect to bitcoind: -28: Loading block index...

IstotaSwiatla, 2021-05-18:
Fresh installation of raspiblitz, after blockchain is fully synced, I'm getting the following on the LCD:

```
               Blockchain Info                                                
│  The bitcoind service is not running.   │                                               
│  Login for more details:                       │                                               
│  Use CTRL+c to EXIT to Terminal      │
```

Restarting makes no difference, after entering password it unlocks the LND wallet ok, but then it hits the same error and after few retries lnd.service gives up with error `(exit after too much restarts/unlocks - restart to try again)`, and the underlying problem in the log: `unable to create chain control: unable to connect to bitcoind: -28: Loading block index...`

Error is identical to [1054](https://github.com/rootzoll/raspiblitz/issues/1054), but there are no issues with the corrupted disk, temp warnings,  etc. No other errors what I can see.

Any ideas what I could try to pinpoint the issue?

`XXdebugLogs.sh` log:
```
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.0
chainnetwork: bitcoin / main
 12:15:38 up 34 min,  2 users,  load average: 1.47, 1.46, 1.33

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-05-18 11:42:24 BST; 33min ago
  Process: 1445 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 1451 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 1472 (bitcoind)
    Tasks: 13 (limit: 4533)
   CGroup: /system.slice/bitcoind.service
           └─1472 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid

May 18 11:42:24 raspberrypi systemd[1]: Starting Bitcoin daemon...
May 18 11:42:24 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Tue 2021-05-18 11:41:16 BST, end at Tue 2021-05-18 12:15:38 BST. --
May 18 11:42:24 raspberrypi systemd[1]: Starting Bitcoin daemon...
May 18 11:42:24 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2021-05-18T11:14:03Z Rolling forward 0000000000000000000a742f471e1cbda74c204ac4c1d586741dc1b8479b4067 (674640)
2021-05-18T11:14:08Z Rolling forward 0000000000000000000d1c350962cfff2efe5e38100e2c12aa8ee957144d8cd7 (674641)
2021-05-18T11:14:12Z Rolling forward 0000000000000000000a12569a5452716e9a5b1a7e78ddfa7ed2c9d9641256a5 (674642)
2021-05-18T11:14:15Z Rolling forward 0000000000000000000b6b0544fd8c54420dafc3943cd9873dd5930292b69dff (674643)
2021-05-18T11:14:18Z Rolling forward 000000000000000000025d9d89100ae8da442f7d45360651aa45b4460bd0b2bd (674644)
2021-05-18T11:14:21Z Rolling forward 0000000000000000000196e7627a43a4980b502b1784963c8eeba2449503c48d (674645)
2021-05-18T11:14:27Z Rolling forward 00000000000000000007383d0e5bd887eb276611b36df13dfbe3a8c80a77861f (674646)
2021-05-18T11:14:32Z Rolling forward 00000000000000000002ad0cc855b74d44b9c813f4ec6b2870eaaf3afaa4a946 (674647)
2021-05-18T11:14:37Z Rolling forward 00000000000000000002b784b9886b222a70c89bf2822cc2f5c0caa951ee546e (674648)
2021-05-18T11:14:43Z Rolling forward 0000000000000000000134474c8c7e5824a13bd1f1183cc3d906875ffa0a6ddb (674649)
2021-05-18T11:14:53Z Rolling forward 0000000000000000000793dc8245a519a8f5b3eb2b6195a55281b031901da669 (674650)
2021-05-18T11:14:59Z Rolling forward 000000000000000000073bb776ee6b9b904a696e049502392e7397524b47f701 (674651)
2021-05-18T11:15:04Z Rolling forward 0000000000000000000c1072dc20b925998b7f6da654ff9bea4706276c508c62 (674652)
2021-05-18T11:15:10Z Rolling forward 00000000000000000005e6ba1a968fb617422d18e1610578c7bdaf6e5e0be19d (674653)
2021-05-18T11:15:14Z Rolling forward 0000000000000000000cc1c048f23556b7b95e368cc5fe5e5a52beb09307d672 (674654)
2021-05-18T11:15:16Z Rolling forward 0000000000000000000a9fa761ecd0e1edfa17d4ee7926ad147698940255a223 (674655)
2021-05-18T11:15:21Z Rolling forward 0000000000000000000b8e8fff850437edd45e8661ffe3cb812549a537d04ae3 (674656)
2021-05-18T11:15:28Z Rolling forward 00000000000000000006ccba0265c03ae9a3b51b66a2f34d23e060f705d8385d (674657)
2021-05-18T11:15:34Z Rolling forward 00000000000000000008e25778fe6b1e432887f4708365d1a41e6331e2c666bf (674658)
2021-05-18T11:15:37Z Rolling forward 00000000000000000005e9b5acac78e9e6c00958f1fe08e38d4912b16cd87aa1 (674659)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
[... 213 lines cut ...]
*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.15'
tempCelsius='51.1'
uptime=2055
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
bitcoinErrorShort='Loading block index'
bitcoinErrorFull='error code: -28 error message: Loading block index...'
startcountLightning=0
lndActive=1
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=0

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999
````

rootzoll, 2021-05-18:
If you see a `Rolling forward` in the logs it seems to recover still on the blockchain part ... give it a day to catch up, the  reboot and let us know if this still has problems.

IstotaSwiatla, 2021-05-19:
Arrrggghhhh should have seen that - indeed it caught up today and it's now up & running
Thank you!

closed 2021-05-19 by IstotaSwiatla