raspiblitz/raspiblitz #2314: Update LNDmanage

rootzoll, 2021-05-18:
There is a new version available - see https://github.com/rootzoll/raspiblitz/issues/2031#issuecomment-840413361 .. it makes sense to update the version in the script for v1.7.1 and then if all is OK and there is further user request re-add it to the SSH menus.

rootzoll, 2022-11-26:
Did update to new verison, install itself worked - but throwing errors when trying to start. WIll keep branch `2314-lndmanage` open but to recheck on later release if nobody fixing this.

```
(venv) admin@192.168.178.147:~/lndmanage ₿ lndmanage
Traceback (most recent call last):
  File "/home/admin/lndmanage/venv/bin/lndmanage", line 5, in <module>
    from lndmanage.lndmanage import main
  File "/home/admin/lndmanage/venv/lib/python3.9/site-packages/lndmanage/lndmanage.py", line 18, in <module>
    from lndmanage.lib.fee_setting import FeeSetter, optimization_parameters
  File "/home/admin/lndmanage/venv/lib/python3.9/site-packages/lndmanage/lib/fee_setting.py", line 12, in <module>
    from lndmanage.lib.forwardings import ForwardingAnalyzer
  File "/home/admin/lndmanage/venv/lib/python3.9/site-packages/lndmanage/lib/forwardings.py", line 9, in <module>
    from lndmanage.lib.node import LndNode
  File "/home/admin/lndmanage/venv/lib/python3.9/site-packages/lndmanage/lib/node.py", line 15, in <module>
    import lndmanage.grpc_compiled.lightning_pb2 as lnd
  File "/home/admin/lndmanage/venv/lib/python3.9/site-packages/lndmanage/grpc_compiled/lightning_pb2.py", line 33, in <module>
    _descriptor.EnumValueDescriptor(
  File "/home/admin/lndmanage/venv/lib/python3.9/site-packages/google/protobuf/descriptor.py", line 755, in __new__
    _message.Message._CheckCalledFromGeneratedFile()
TypeError: Descriptors cannot not be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).

More information: https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates
```

rootzoll, 2022-12-24:
TODO: merge #3575 & test for v1.9.0rc2

rootzoll, 2023-05-10:
Updated lndmanage  to v0.15.0 - installs OK but I run on my test machine into timeouts.
This should be tested by community when 1.9.0RC5 or even after final 0.19.0 release.

rootzoll, 2024-04-23:
Check if new release works:
https://github.com/bitromortac/lndmanage/releases/tag/v0.16.0

geco91, 2024-06-04:
lndmanage 0.16.0 looks good - open PR #4612 - recommend for final testing

PR #4612 (2024-06-09): #2314 Update lndmanage to v0.16.0
fixes #2314
closed 2024-06-09 by rootzoll