raspiblitz/raspiblitz #2321: apt upgrade

dalamanster, 2021-05-22:
Hi there,

is it safe to update raspiblitz with "apt update + apt upgrade" ? 
Or should I always wait for official raspiblitz releases ? 

I want to have packages with latest security updates.

Thank you

openoms, 2021-05-24:
we can't predict if new versions of everything will break things.
If there is anything specific you want to update use
`sudo apt install PACKAGE_NAME`

If the system breaks the tested release (latest SDcard image) is there as a fallback.

closed 2021-05-24 by dalamanster

openoms, 2022-10-12:
Since the Tor configuration has been refactored apt upgrade is less prone to break and it is the besst practice running linux to keep paackages up to date.
Usually it is not necessary to run this between releases, but can do especially if there is some known vulnerability.