raspiblitz/raspiblitz #2325: Electrum server: After update, cert chganges

peterflock, 2021-05-25:
After the update of Raspi, I was not able to connect to Electrum server over Tor. It turns out that it was because cert changed:
```
W/i | interface.[XXXXXXXXXXXXXXXXXXXXXXXXXXX.onion:50002] | Cannot connect to main server due to SSL error (maybe cert changed compared to "/home/<user>/.electrum/certs/XXXXXXXXXXXXXXXXXXXXXXXXXXX.onion"). Exc: ConnectError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1091)'))
```

Is such a change needed? 

(Removing `/home/<user>/.electrum/certs/XXXXXXXXXXXXXXXXXXXXXXXXXXX.onion` solved the issue.)

nyxnor, 2021-05-25:
you will use port 50001 with Tor. 50002 is not needed.

peterflock, 2021-05-25:
When using 50001 it doesn't seem to work
```
I/n | network | connecting to XXXXXXXXXXXXXXXXXXXX.onion:50001:s as new interface
I/i | interface.[XXXXXXXXXXXXXXXXXXXX.onion:50001] | disconnecting due to: ConnectError(ConnectionResetError())
I/n | network | couldn't launch iface XXXXXXXXXXXXXXXXXXXX.onion:50001:s -- CancelledError()

nyxnor, 2021-05-25:
What is the other software you are using? This hasnt happened to me with mobile wallets and electrum.

peterflock, 2021-05-26:
This is Electrum. I have found a problem. I was connection using
`--server XXXXXXXXXXXXXXXXXXXXX.onion:50001:s` instead of  `--server XXXXXXXXXXXXXXXXXXXXX.onion:50001:t` (s/t)

Now `50001` port works.

peterflock, 2021-05-26:
But out of curiosity, why is the difference? One is (50002) HTTPS and the other (50001) HTTP right? WHy is HTTPS not needed when using Tor? Whats the difference from securoty perspective?

nyxnor, 2021-05-26:
>But out of curiosity, why is the difference? One is (50002) HTTPS and the other (50001) HTTP right? Why is HTTPS not needed when using Tor? Whats the difference from security perspective?

https://community.torproject.org/onion-services/advanced/https/

When visiting a site over the onion services protocol, the Tor protocol prevents data in transit from being read or manipulated by man in the middle attacks, and the onion service protocol validates that the user is connected to the domain name in the browser address bar. No certificate authority is required for this proof, because that name is the actual public key used to authenticate the underlying connection.

closed 2021-05-28 by peterflock