raspiblitz/raspiblitz #2326: Stuck rebooting trying to recover

Svobodan, 2021-05-25:
After an internet outage for half a day raspiblitz 1.6.3 is trying to recover.

[image]

It rebooted about 2 or 3 times then asked to set new Password A after which rebooted again doing recovery.
Now it does not even boot at all only showing white screen.

Is there a way to fix it without loosing opened channels?

nyxnor, 2021-05-25:
Looks hardware issue. EXT4 is the file system on the ssd. Either way, try to flash the sd card again and/or use a new sd card as they wear out rapidly.

Svobodan, 2021-05-26:
Even with a new SD card flashed I am getting the same result. Still keeps rebooting (with the same EXT4-fs error) and "recovering". Does that mean that I likely have an SSD failing?

nyxnor, 2021-05-26:
Possible, but more probable is the enclosure. SSDs are reliable, bad enclosures are common.
Try a recommended enclosure, as it is cheaper than a new disk, then if it still does not work, you will need a new disk.

Svobodan, 2021-05-26:
Alright, thanks for your help. I hope you are right.

nyxnor, 2021-05-28:
pls close this issue,

closed 2021-05-28 by Svobodan