raspiblitz/raspiblitz #2327: IPv6 and TOR not useable at the same time

janis-mueller, 2021-05-26:
Recently I tested out to switch on the IPv6 option via the Raspiblitz internet.sh scripts, which worked perfectly without  any problems but in order for it to work I had to disable TOR first.
When I tried to keep TOR enabled and switch on IPv6 to make my node reachable via my public IP address and TOR, none of the services listened via tcp6 when I checked it with netstat, even though the scripts reported a success. 

Would be very useful to have the node reachable via TOR and IPv6 but unfortunately this doesn't work so far.

nyxnor, 2021-05-26:
https://github.com/rootzoll/raspiblitz/issues/2209#issuecomment-830614032
It is marked for 1.8.1, but you can try to get it working earlier and share your experience.

janis-mueller, 2021-05-27:
Thanks, I see the duplicate issue, closing.

closed 2021-05-27 by janis-mueller