raspiblitz/raspiblitz #2329: Unknown problem - please advise!

mh-ash, 2021-05-26:
I’m running into an error today where my channels are not seeming to backup and I am getting err 400 a lot and when using BOS I’m getting “balanceofsatoshis update check failed”. My most recent channel is also not showing up on the explorers. Please help if you know what could be the problem or have had this same issue!

Other notes - I'm getting err 500 in RTL, and I cannot send out ln transactions from my node. 

I'm not technical and don't know where to look or what to look for. I'm happy to paste code if someone can direct me to which code I should be looking at. 

Thanks in advance! Don't want my node to fail.

mh-ash, 2021-05-26:
I have the debug report now and it is really long but I don't know what parts I need to look at.

nyxnor, 2021-05-26:
to find channel backups on the device:
`sudo find / -name channel.backup`
enter the folders with `cd folderName`
`ls -l` to list folder with time it was last modified.

Look at LND part in the debug logs.

mh-ash, 2021-05-26:
> to find channel backups on the device:
> 
> `sudo find / -name channel.backup`
> 
> enter the folders with `cd folderName`
> 
> `ls -l` to list folder with time it was last modified.
> 
> 
> 
> Look at LND part in the debug logs.

nyxnor, 2021-05-26:
No more folders if you wait some time?
I got:
>/home/admin/backups/scb/channel.backup
/boot/channel.backup
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup

mh-ash, 2021-05-26:
> No more folders if you wait some time?
> 
> I got:
> 
> >/home/admin/backups/scb/channel.backup
> 
> /boot/channel.backup
> 
> /mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup

Just checked and yes - 

/home/admin/backups/scb/channel.backup
/boot/channel.backup
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup
/mnt/hdd/backup_lnd/data/chain/bitcoin/mainnet/channel.backup

nyxnor, 2021-05-26:
Enter those folders with `cd folderName`
example:
`cd /mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup`
`ls -l`
Can you see the backups?

mh-ash, 2021-05-27:
> Enter those folders with `cd folderName`
> 
> example:
> 
> `cd /mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup`
> 
> `ls -l`
> 
> Can you see the backups?

I think so (a bunch of scripts came up).  In every one of them lndrescue.... is in red.

mh-ash, 2021-05-27:
> > Enter those folders with `cd folderName`
> > 
> > example:
> > 
> > `cd /mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup`
> > 
> > `ls -l`
> > 
> > Can you see the backups?
> 
> I think so (a bunch of scripts came up).  In every one of them lnd-rescue.... is in red. 

I'm also frequently seeing things that say I'm out of space on my device, but I can't figure out why or how. I feel like something is very wrong, but don't know what it is.

nyxnor, 2021-05-27:
to see available space (disk free) with human readable format (-h)
`df -h`

mh-ash, 2021-05-27:
> to see available space (disk free) with human readable format (-h)
> `df -h`

[image]

nyxnor, 2021-05-27:
How many services are you using?
To find largest files/folder in linux https://www.cyberciti.biz/faq/how-do-i-find-the-largest-filesdirectories-on-a-linuxunixbsd-filesystem/

`du -a / | sort -n -r | head -n 10`

openoms, 2021-05-27:
@mh-ash this looks weird, likely because of a broken SDcard. 
Get  a new card (see the shopping list) and reflash the latest image.

HazeMero, 2021-05-28:
Once I had the same problem, node was hardly reachable and acting weird. Turned out some logfiles went crazy (15 GB):
[image]

After deleting the big files everything went back to normal. I couldn't find out where the big logs came from.

Logrotate configs:
[image]

[image]

mh-ash, 2021-05-28:
Yes, that is what seems to have happened (from my limited understanding). @nyxnor helped me to clear it away and now my CPU load has dropped from ~5.7 etc to 0.5, and I have space on my drive too!  It's also running like 40 degrees F cooler now. 

I don't know how it happened, but grateful that it is back running normal.

nyxnor, 2021-05-30:
There was something stuck there and mh-ash reached for me on keybase. Logs were 15GB and restart did also cleaned more space on the sd card. I should have asked what was the biggest logs, but I just said to delete them. For now @mh-ash, close this, if it happens again, reopen.

@HazeMero Those `messages.1` with 15GB appears to be boot logs, probably looped on boot and stucked there. If they are still big there, just delete it.

HazeMero, 2021-05-30:
Yes deleted everything also. Just thought it helps if someone wants to dive deeper in this issue and find out where the big logs came from. Several logfiles grew to 15 GB - messages, syslog and user.log

closed 2021-06-01 by mh-ash