raspiblitz/raspiblitz #2330: LND Balance changing without new TXNs

inpharmaticist, 2021-05-26:
Hello,

I've noticed my lightning balance changing every now and then without doing any transactions. In this case, it went up ~11k sats. Not sure what's going on.

[image]
[image]

inpharmaticist, 2021-05-26:
Just went up another 10k sats

[image]

openoms, 2021-05-27:
This is normal and changiing with the mempool (`bitcoin-cli estimatesmartfee 1` results).

There is always a reserve on the channels you opened to be able to pay a force (unilateral) close.
As miner fees go up, the reserve goes up as well and vice versa.

This reserve is on;y used in case of a force close and the sats are yours, but cannot be spent on LN when the miner fees are high.

inpharmaticist, 2021-05-27:
Wow, I'm surprised I've never noticed before. Is there a way to monitor this on the lncli side, perhaps as it changes over time?

closed 2021-05-27 by inpharmaticist