raspiblitz/raspiblitz #2333: Problems with Nginx Startup, "Give root password for maintenance or press Control-D to continue" and unwanted Hardware Test

HazeMero, 2021-05-28:
My Node is running ok but i ran into several problems I couldn't solve:

1. When rebooting it always stucks with "Give root password for maintenance or press Control-D to continue"
Corresponding with #782 I tried changing USB Port and enabling the Auto Unlock Feature for LND. Didn't fix the issue next step is upgrading to 1.7 with a new SD-Card but at the moment I don't have one spare. So this is likely a SD-Card problem according to #1053 but I dont know how the verify.

2. Since some time my nginx failed to spin up at start like stated in the debug report. sudo nginx -t tells me everything is ok. A simple sudo service nginx start solves the problem, while I still don't know why it fails in the first try after reboot.

3.  The newest issue just occured a few days ago: everytime the node restarts it does a Hardware Test and acts as if its started from scratch - showing options which kind of node i wanna start. With cancel I get back to CLI where everything is looking fine (after i start nginx manually) bitcoin and lnd is running and working. Is there a flag or something for HardwareTest that could be stuck?

4. raspberrypi systemd[1]: bitcoind.service: Failed to parse PID from file /home/bitcoin/.bitcoin/bitcoind.pid: Invalid argument
According to #1901 I checked both PID Files /home/bitcoin/.bitcoin/bitcoind.pid and /mnt/hdd/bitcoin/bitcoind.pid. They show the correct PID but are only accessible with sudo not sure if intended?

My debug report: https://termbin.com/j13l

I'm pretty new to posting issues, so if I should split these or did something wrong just let me know I will try to fix it. Posted them together because they maybe related (SD/HDD failure?).

Thanks for your time, attention and all the work you already put into this awesome project!

nyxnor, 2021-05-28:
Good that you made an organized issue.
1. you can flash the same microSD card, it is just better to use a new one. there was only one person that reported to me that had this password maintenance issue with 1.7, and he was using a bigger display.
2. See nginx logs. Open that file and delete the duplicate in line 5.
>nginx: [emerg] a duplicate listen [::1]:23002 in /etc/nginx/sites-enabled/btcpay_tor.conf:5
3. Hardware failure or incorrect configuration files.
4. This issue will be continued there if more people report it, but you are still on 1.6.3

Recommended: flash the v1.7.0 and see if everything fixes, else, continue posting on this topic here.

HazeMero, 2021-05-31:
Thanks, everything works fine with v1.7.0

closed 2021-05-31 by HazeMero