raspiblitz/raspiblitz #2335: unify services variables - discussion

nyxnor, 2021-05-29:
Is there an explicit reason for this to be different? Some are easy to follow, thunderhub is just the t, but the services above are very distinct.

script=bonus.btcpayserver.sh
hiddenservice=btcpay
raspiblitz.conf=BTCPayServer

script=bonus.rtl.sh
hiddenservice=RTL
raspiblitz.conf=rtlWebinterface

Can't rtl be always rtl in every mention?

openoms, 2021-06-01:
Good point. Should use low caps and the same name as in the script filename.
RTL is due for a major overhaul with clightning and the parallel services so can parse the config file as well.
will have prefixes for the network ( mainnet = '', testnet = 't', signet = 's' ) and 'c' for clightning so will have the options of `rtl`, `trtl`, `crtl`, `tcrtl` and `scrtl` in the configfile.

Should ake the systemd service name more consistent as well (eg drop the caps for RTL). 
Some will have the `d` in the end anyway so can't do it with all (like `litd`, `loopd`, `poold` etc)

openoms, 2021-08-23:
Renamed `cryptoadvance-specter` to `specter` and  `Specter Desktop``in the menus for better discoverability and shorter service name in 7c329f85

nyxnor, 2021-08-24:
Which line did you remove the old entry in raspiblitz.conf? just to make it clean.

openoms, 2021-08-24:
@nyxnor did bot remove any. The conf entray was `specter=` before and remains so.

rootzoll, 2021-09-09:
> Can't rtl be always rtl in every mention?

Thinks like that are because it grown not over time - looking back it makes sense to hamonize it. Now that RC1 is out for v1.7.1 I am open for someone doing it - not just high on my personal prio.

Changes on services are not a big deal - because we just need to make them consistent thru the code and it wil play well with an update on the user side. Changes in `raspiblitz.conf` need also an update path.