raspiblitz/raspiblitz #2337: Electrum server via Tor is not returning up-to-date data..?

Overtorment, 2021-05-29:
Chasing a bug described here:  https://github.com/BlueWallet/BlueWallet/issues/3210

quote:

> I noticed when opening Bluewallet configured to use my own raspiblitz electrum server via tor it doesnt refresh the wallet (shows no new transactions) no  matter how long i wait.
> But when i open my Phoenix Wallet which is also connected to the Electrum server right before opening Bluewallet it instantly gets the new transactions  and everything works as intended.

frankly, this sounds very weird to me and I have no idea where to start debugging.

nyxnor, 2021-06-07:
https://github.com/BlueWallet/BlueWallet/issues/3210#issuecomment-856157773

Overtorment, 2021-07-24:
a clue:

```
echo  -e -n '{"id": 1, "method": "server.version", "params": ["bluewallet", "1.4"]}\n{"id": 2, "method": "blockchain.headers.subscribe", "params": []}\n' | nc oq3bcnn3ceyg4efmq3wed24n363xla7nf6esj2cc4zxez6wauhwb7wyd.onion 50001 -i 1 -x 127.0.0.1:9050

nc: connection failed, SOCKSv5 error: TTL expired
```

^^^ sometimes it works sometimes not.

my guess is that something is wrong with Tor

nyxnor, 2021-07-24:
@exitfiat

exitfiat, 2021-07-24:
I updated tor on my raspiblitz and restarted it. I could refresh the wallet in bluewallet but i dont know if this is a fix or it stops working again sometimes.
Will watch it, mayve the problem solves itself.

openoms, 2022-07-04:
closing for inactivirty

closed 2022-07-04 by openoms