raspiblitz/raspiblitz #2339: Fully Noded does not find node

Lobbelt, 2021-06-03:
After connecting Fully Noded with the QR code to my Raspiblitz (on a RPi4), Fully Noded still complained "Looks like your node is not on, make sure it is running and try again."

I have tried rebooting the Raspiblitz, does not help.

For what it's worth - ElectRS is still building the index, but not sure whether this has anything to do with this issue.

openoms, 2021-06-03:
v1.7?
Similar was fixed here: https://github.com/rootzoll/raspiblitz/pull/2177/files

try running:
`/home/admin/config.scripts/internet.hiddenservice.sh bitcoin8332 8332 8332`
to fix

Lobbelt, 2021-06-03:
I ran that command, didn't work - rebooted the Raspiblitz, which also didn't help.

Afterwards I removed the connection in FN and re-connected using the QR code again. Now it no longer says the node is not on, but "The network connection was lost."

(Apologies - I'm not super tech savvy so I may be missing crucial things here)

openoms, 2021-06-04:
Not sure what the issue is. Someone with an iOS device should test.

Lobbelt, 2021-06-05:
Some more background: I managed to connect Fully Noded to the RB by entering the onion address (as displayed above the QR code when connecting), rpcuser and rpcpassword manually. Connection now works.

Not sure if that's what went wrong, but I noticed the onion address is different from the one that Fully Noded read from the QR code - the one that didn't work (even after 5 tries).

openoms, 2021-06-13:
Fixed commands to get right QR:

```
source /mnt/hdd/raspiblitz.conf
RPC_USER=$(sudo cat /mnt/hdd/bitcoin/bitcoin.conf | grep rpcuser | cut -c 9-)
PASSWORD_B=$(sudo cat /mnt/hdd/bitcoin/bitcoin.conf | grep rpcpassword | cut -c 13-)
hiddenService=$(sudo cat /mnt/hdd/tor/bitcoin/hostname)
quickConnect="btcstandup://${RPC_USER}:${PASSWORD_B}@${hiddenService}:8332/?label=${hostname}"
qrencode -t UTF8 $quickConnect
```

rootzoll, 2021-10-11:
Closing for final release.

closed 2021-10-11 by rootzoll