raspiblitz/raspiblitz #2341: FAIL: bitcoin failed to start

crazy999666676, 2021-06-04:
Hello, 

I copied the "bitcoin" folder from my Raspiblitz. With "sudo cp (source) (target) -a -r -v"
Then I copied the "bitcoin" folder to a new SSD.
Unfortunately I can't get any further with the installation.
The SD card has been flashed and has the version Raspiblitz 1.7
A log file is attached.
Thank you for your support!

[Error_Log_Raspi.odt](https://github.com/rootzoll/raspiblitz/files/6598598/Error_Log_Raspi.odt)

d-hoffi, 2021-06-07:
I am using [Raspibolt](https://stadicus.github.io/RaspiBolt/) so I am not 100% sure if my idea works. Unfortunately I don't have an empty disk to try it out myself.

1. First of all you need to stop all services or other applications with `$ sudo systemctl stop PROCESSNAME`
2. Than unplug all USB devices/disks. 
3. We need some information about your drive from the kernel. So first we clear the kernel buffer with `$ sudo dmesg -C` and than we wait for new messages with `$ sudo dmesg -w` 
4. After running the last command you can connect your new external disk. The system will detect it automatically and displays the details from the `dmesg` command. The output should look similar to this:
```
[  726.547907] usb 2-1: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[  726.579304] usb 2-1: New USB device found, idVendor=152d, idProduct=0578, bcdDevice= 3.01
[  726.579321] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  726.579333] usb 2-1: Product: USB 3.0 Device
[  726.579346] usb 2-1: Manufacturer: USB 3.0 Device
[  726.579357] usb 2-1: SerialNumber: 000000005B3E
[  726.582254] usb 2-1: UAS is blacklisted for this device, using usb-storage instead
[  726.582350] usb 2-1: UAS is blacklisted for this device, using usb-storage instead
[  726.582364] usb-storage 2-1:1.0: USB Mass Storage device detected
[  726.582674] usb-storage 2-1:1.0: Quirks match for vid 152d pid 0578: 1800000
[  726.582783] scsi host0: usb-storage 2-1:1.0
[  727.598422] scsi 0:0:0:0: Direct-Access     INTENSO  SATA III SSD     0301 PQ: 0 ANSI: 6
[  727.599182] sd 0:0:0:0: Attached scsi generic sg0 type 0
[  727.605796] sd 0:0:0:0: [sda] 937703088 512-byte logical blocks: (480 GB/447 GiB)
[  727.606519] sd 0:0:0:0: [sda] Write Protect is off
[  727.606536] sd 0:0:0:0: [sda] Mode Sense: 47 00 00 08
[  727.607982] sd 0:0:0:0: [sda] Disabling FUA
[  727.607998] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[  727.611337]  sda: sda1
[  727.614890] sd 0:0:0:0: [sda] Attached SCSI disk
```
5. We take a look at the second line. Safe your `idVendor` and your `idProduct` (in my example 152d and 0578) into a textfile. Than exit `dmesg` with `Ctrl` - `C`
6. List all block devices with additional information and make a note of the partition you want to use (in my case "sda1"):
```
$ lsblk -o NAME,MOUNTPOINT,UUID,FSTYPE,SIZE,LABEL,MODEL
> NAME        MOUNTPOINT UUID                                 FSTYPE   SIZE LABEL  MODEL
> sda                                                                447.1G        SATA_III_SSD
> └─sda1                 9ec0b784-d448-4757-a3b2-8abd57c544f3 ext4   447.1G
> mmcblk0                                                             14.9G
> ├─mmcblk0p1 /boot      5203-DB74                            vfat     256M boot
> └─mmcblk0p2 /          2ab3f8e1-7dc6-43f5-b0db-dd5759d51d4e ext4    14.6G rootfs
```
If your disk doesn't have any partitions follow this [guide](https://superuser.com/questions/643765/creating-ext4-partition-from-console) to create one. Make a note of the right `UUID`. We need the `UUID` for `sda1` in my case.
7. Edit the `fstab` file and add the following as a new line at the end, replacing `123456` with your own `UUID`:
```
$ sudo nano /etc/fstab
UUID=123456 /mnt/ext ext4 rw,nosuid,dev,noexec,noatime,nodiratime,auto,nouser,async,nofail 0 2
```
8. Create a directory on that disk. On Raspibolt this directory is `/mnt/ext/` but I am not sure what it is at Raspiblitz:
`$ sudo mkdir /mnt/ext`
9. Mount all drives and check the file system. Is “/mnt/ext” listed?
```
$ sudo mount -a
$ df -h /mnt/ext
> Filesystem      Size  Used Avail Use% Mounted on
> /dev/sda1       440G   73M  417G   1% /mnt/ext
```
10. Set the owner to the bitcoin user. `$ sudo chown -R bitcoin:bitcoin /mnt/ext/`. Switch to user “bitcoin”, navigate to the external drive and create the bitcoin directory:
```
$ sudo su - bitcoin
$ cd /mnt/ext
$ mkdir bitcoin
$ ls -la
> total 28
> drwxr-xr-x 4 bitcoin bitcoin  4096 Dec 12 17:43 .
> drwxr-xr-x 4 root    root     4096 Dec 12 17:38 ..
> drwxr-xr-x 2 bitcoin bitcoin  4096 Dec 12 17:43 bitcoin
> drwx------ 2 bitcoin bitcoin 16384 Dec 12 17:30 lost+found
```
11. shutdown your Pi and copy the data you need to the disk using a Desktop PC. You could copie your data before point 1 aswell. 

I hope this will do it. Hopefully the location of you fstab etc is the same. Maybe you need to change the directory names a bit that it fits for the raspiblitz.

openoms, 2022-01-06:
This should be solved with v1.7.1. Closing for inactivity.

closed 2022-01-06 by openoms