raspiblitz/raspiblitz #2342: nginx failed - electrum

Adrianvic032, 2021-06-04:
Hi there,
  Please advice. The following error is up when I try to start Electrum server on newly installed fully syncd Raspiblitz v: 1.7.1.

Error:
Nginx:(emerg) PEM_read_bio_DHparams("/etc/ssl/certs/dhparams.pem) failed SSL error.

nginx config file: /etc/nginx/nginx.conf  -  ssl error : 0909006c

Any help - highly appriciated - Thanks

openoms, 2021-06-04:
there is no v1.7.1 yet.
Plase reflash the latest v1.7.0 image.

Adrianvic032, 2021-06-04:
Sorry, typo error. Its Raspiblitz version 1.7.0
Thanks for your promt email reply. I still get 'nginx failed' error and
electrum is not starting. Please advice. Thanks

Adrianvic032, 2021-06-05:
Sorted after re-image from scratch. Thanks

nyxnor, 2021-06-06:
Good that it was sorted. Please close this Adrianvic

closed 2021-06-14 by rootzoll