raspiblitz/raspiblitz #2343: HTTPS not working after running letsencrypt

ChrisSwiz, 2021-06-05:
Hello, I'm testing an online shop setup with woocommerce and btcpayserver and want to use IP2TOR with Letsencrypt HTTPS Domain for public access to my home btcpayserver behind TOR.

Subscribing to IP2TOR bridge went fine, then I tried to install the cert for a duckdns subdomain but during the very short script run in the terminal I can see how there is an error 'curl tor proxy not working' (see attachment). Strangely, I get a success message afterwards but the duckdns subdomain still has the old self-signed certificate, the new cert was obviously not installed properly.

I then tried to use dynu.com and have the same problem. The cert just does not get installed properly. With Dynu.com I at least get a message that the cert install failed.

With regards to the reason I can only speculate. Before I ran certbot I had an update of Tor from source code. I hope that did not mess up the Tor configuration. It looks to me as if certbot cannot run properly due to a problem with tor.

I have my own domain with DNS settings on cloudflare and would like to install my own certificate with certbot but how do I do this in the raspiblitz install and file structure? 

**Can someone please write up how to install your own domain's cert on raspiblitz?** 

[image]

Thank you very much!

rootzoll, 2021-06-14:
Which version of RaspiBlitz are you running?

I added the some lables to the issue so that others may better find it to contribute a tutorial on using a own domain cert.

ChrisSwiz, 2021-06-15:
I am running Raspiblitz v1.7.0 . Thank you for labeling, it would be great if someone was to put such a tutorial together. I'm sure there are many people out there who would like to use Raspiblitz BTCPayserver install with their own domain.

ChrisSwiz, 2021-07-23:
I tried the setup again with another domain and after patching raspiblitz and it is working now with the duckdns subdomain and the HTTPS installed fine. I am very happy !

closed 2021-07-23 by ChrisSwiz