raspiblitz/raspiblitz #2346: https://openoms.gitbook.io/raspiblitz/alternative.platforms/dietpi/odroid_hc1_xu4

rek79, 2021-06-06:
Hi,
I was hoping someone here could verify this site and image is legit.  I landed on the referenced site and installed prior to finding this official github site.   Apologies in advance if there is a better place to post this question.  I wasn't sure were else to ask.

Site Address:
https://openoms.gitbook.io/raspiblitz/alternative.platforms/dietpi/odroid_hc1_xu4
Image:
sha256sum RaspiBlitz1.0_OdroidHC1_DietPi6.21.1.img: 96ee8700f52a12fb5b13fab3fffcdcf5d364c6dd16c580d969e421cef2cd7cc2

nyxnor, 2021-06-07:
It is legit (from @openoms), it is just outdated, Blitz v1.0, now we are currently at v.1.7.
If you want to use DietPI, build [your image](https://openoms.gitbook.io/raspiblitz/alternative.platforms/dietpi/odroid_hc1_xu4#build-your-own-dietpi-image) and run the [build script](https://openoms.gitbook.io/raspiblitz/alternative.platforms/dietpi/odroid_hc1_xu4#run-the-raspiblitz-build_sdcard-sh-script).

rek79, 2021-06-07:
Thanks for confirming!  So, the boot loop that occurs when updating DietPi is the main reason I will need to reflash?  I was hoping it would update to 1.7.  I'm almost done with node sync (took about 1 week on Odroid HC2).  Torrent and other methods didn't work.

nyxnor, 2021-06-07:
What message does it show for this boot loop?

rek79, 2021-06-07:
I was referencing the boot loop issue in your link above "your image".  Reading further, it seems this was only a problem while attempting an update from DietPi v6.14.  The image I flashed onto my Ordiod is using DietPi v6.21.1 so hopefully I will not have that problem.  I'm currently waiting for sync at 91.65%.
[image]
[image]

rek79, 2021-06-07:
I'm reading between the lines and assuming the safest bet is to follow your original guidance of re-flashing to latest version 1.7.  Is it possible to keep the attached hard drive intact, or will the re-flash trigger a reformat of the hard drive?

openoms, 2021-06-07:
I had better success with the Armbian images, but they are neither those are tested for the latest versions.

Also keep in mind that the Odroid XU4 is 32 bit, so LND will stop working when the channel.db reaches 1GB (meaning that don't plan using at as a busy routing node for long).

Implementing the setup process will likely involve manual steps where you need to go in the command line and edit some scripts yourself.

rek79, 2021-06-07:
I'm not afraid to hack around with the scripts, and enjoy doing so.  However, it sounds like I should just bite the bullet and migrate to the 64-bit architecture.  I may give it a shot for the learning experience anyways but will try not to bog you guys down with questions.  Thanks for taking the time to lead me down the right path.  You can close this out if there isn't anything else to add, cheers!

openoms, 2021-06-09:
Running:
```
# download the build script
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.7/build_sdcard.sh
# inspect the script 
cat build_sdcard.sh
# run with sudo or from root
bash build_sdcard.sh false false rootzoll v1.7 headless true US
```

worked perfectly for me.
There are a couple of options to choose from as I did above:

```
root@odroidxu4:~# bash build_sdcard.sh

*****************************************
* RASPIBLITZ SD CARD IMAGE SETUP v1.7   *
*****************************************
For details on optional parameters - see build script source code:
1) will use NO-INTERACTION --> 'false'
2) will use FATPACK --> 'false'
3) will use GITHUB-USERNAME --> 'rootzoll'
4) will use GITHUB-BRANCH --> 'dev'
5) will use DISPLAY-CLASS --> 'hdmi'
6) will use TWEAK-BOOTDRIVE --> 'true'
7) will use WIFI --> 'US'
X) will use CPU-ARCHITECTURE --> 'arm'
X) will use OPERATINGSYSTEM ---> 'armbian'
Do you agree with all parameters above? (yes/no) no
root@odroidxu4:~# bash build_sdcard.sh false false rootzoll v1.7 headless true US
```

notes:
The command to expand the SDcard filesystem in Armbian is:
`sudo /usr/lib/armbian/armbian-resize-filesystem start`

Might release an SDcard image for this 32bit board with c-lightning for testing.

rek79, 2021-06-10:
@openoms Thanks for the clarification on build options.  The image worked great and build succeeded.  However, I think my previous config saved on the HDD caused some issues.  Bitcoind service was not detected.  It may be that my pw's previously set now conflict with the default passwords.  There was no prompt to set Password A, Password B, etc...  I ran the "repair" command and deleted HDD data except for blockchain.  That also did not seem to work.

Here is one warning I received during build:
[image]

This is the message I receive after boot process checks for bitcoind and prompts for Bitcoin setup:
[image]

openoms, 2021-06-10:
to be clear:
build the SDcard first without the disk attached and reboot with a clean disk attached or a try your previous.
There should be only one partition on the disk with an EXT4 filesystem and labeled `BLOCKCHAIN`.

rek79, 2021-06-24:
So I tried completely formating the external drive and starting over with fresh image and get the same result.  The only warning I see in the install log is this "mv: cannot stat './raspiblitz.info': No such file or directory".  See below for more context:

*** ADDING MAIN USER admin ***
Adding user `admin' ...
Adding new group `admin' (1002) ...
Adding new user `admin' (1002) with group `admin' ...
Creating home directory `/home/admin' ...
Copying files from `/etc/skel' ...
Adding user `admin' to group `sudo' ...
Adding user admin to group sudo
Done.
%sudo ALL=(ALL) NOPASSWD:ALL
mv: cannot stat './raspiblitz.info': No such file or directory

rek79, 2021-06-24:
Full install log:
`root@odroidxu4:~/download# wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.7/build_sdcard.sh
--2021-06-23 20:35:17--  https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.7/build_sdcard.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133, 185.199.109.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 40027 (39K) [text/plain]
Saving to: ‘build_sdcard.sh’

build_sdcard.sh                                     100%[=================================================================================================================>]  39.09K  --.-KB/s    in 0.006s

2021-06-23 20:35:17 (5.92 MB/s) - ‘build_sdcard.sh’ saved [40027/40027]

root@odroidxu4:~/download# bash build_sdcard.sh false false rootzoll v1.7 headless true US

*****************************************
* RASPIBLITZ SD CARD IMAGE SETUP v1.7   *
*****************************************
For details on optional parameters - see build script source code:
1) will use NO-INTERACTION --> 'false'
2) will use FATPACK --> 'false'
3) will use GITHUB-USERNAME --> 'rootzoll'
4) will use GITHUB-BRANCH --> 'v1.7'
5) will use DISPLAY-CLASS --> 'headless'
6) will use TWEAK-BOOTDRIVE --> 'true'
7) will use WIFI --> 'US'
X) will use CPU-ARCHITECTURE --> 'arm'
X) will use OPERATINGSYSTEM ---> 'armbian'
Do you agree with all parameters above? (yes/no) yes
Building RaspiBlitz ...

*** INSTALL TOR BY DEFAULT ***

Reading package lists... Done
Building dependency tree
Reading state information... Done
dirmngr is already the newest version (2.2.12-1+deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
*** Adding KEYS deb.torproject.org ***
gpg: keybox '/root/.gnupg/pubring.kbx' created
gpg: key EE8CBC9E886DDD89: 83 signatures not checked due to missing keys
gpg: /root/.gnupg/trustdb.gpg: trustdb created
gpg: key EE8CBC9E886DDD89: public key "deb.torproject.org archive signing key" imported
gpg: Total number processed: 1
gpg:               imported: 1
gpg: no ultimately trusted keys found
OK
- OK key added
*** Adding Tor Sources to sources.list ***
torSourceListAvailable=0
- adding TOR sources ...
- using https://deb.torproject.org/torproject.org buster
deb https://deb.torproject.org/torproject.org buster main
deb-src https://deb.torproject.org/torproject.org buster main
- OK sources added
*** Install & Enable Tor ***
Get:1 http://security.debian.org buster/updates InRelease [65.4 kB]
Get:2 http://deb.debian.org/debian buster InRelease [122 kB]
Get:3 http://deb.debian.org/debian buster-updates InRelease [51.9 kB]
Get:4 http://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Get:6 http://security.debian.org buster/updates/main armhf Packages [360 kB]
Get:7 https://deb.torproject.org/torproject.org buster InRelease [3,524 B]
Get:5 https://armbian.tnahosting.net/apt buster InRelease [18.4 kB]
Get:8 http://deb.debian.org/debian buster/main armhf Packages [10.5 MB]
Get:9 http://deb.debian.org/debian buster-updates/main armhf Packages.diff/Index [7,624 B]
Get:10 http://deb.debian.org/debian buster-updates/main armhf Contents (deb).diff/Index [7,132 B]
Get:11 http://deb.debian.org/debian buster-updates/main armhf Packages 2021-06-21-1401.46.pdiff [448 B]
Get:12 http://deb.debian.org/debian buster-updates/main armhf Packages 2021-06-23-1401.37.pdiff [4,287 B]
Get:13 http://deb.debian.org/debian buster-updates/main armhf Contents (deb) 2021-06-21-1401.46.pdiff [152 B]
Get:14 http://deb.debian.org/debian buster-updates/main armhf Contents (deb) 2021-06-23-1401.37.pdiff [387 kB]
[... 1592 lines cut ...]
c5cc30e1edbc62592dab95d5d03cfa1fc4d172120a20054ab4085bf032bf2402
Downloaded binary SHA256 checksum: e79200b6db1b3030f66be4cebe156fd1d6b1551dda64a9b0739d0ec68aacf212

****************************************
OK --> VERIFIED LND CHECKSUM IS CORRECT
****************************************

- install LND binary
- OK install of LND done

*** raspiblitz.info ***
baseimage=armbian
cpu=arm
displayClass=headless

**********************************************
BASIC SD CARD BUILD DONE
**********************************************

Your SD Card Image for RaspiBlitz is ready (might still do display config).
Take the chance & look thru the output above if you can spot any errors or warnings.

IMPORTANT IF WANT TO MAKE A RELEASE IMAGE FROM THIS BUILD:
1. login fresh --> user:admin password:raspiblitz
2. run --> ./XXprepareRelease.sh`

rek79, 2021-06-24:
Here is where I'm stuck now:
[image]

rek79, 2021-06-24:
Update:
I unmounted the drive with:
`sudo umount /mnt/hdd`

Then I ran:
`.\30initHDD.sh`

Now it seems to be progressing.  I'm currently on:
`*** Start bitcoin ***
- This can take a while ..
Created symlink /etc/systemd/system/multi-user.target.wants/bitcoind.service → /etc/systemd/system/bitcoind.service.
`
My hunch is there is still some file permissions not being set correctly if the build script detects an already formatted ext4 storage drive.

openoms, 2021-06-24:
dump logs in pastebin rather than in comment to remain readable.

I got one working based in armbian, but needed to comment out parts of the scripts looking for the IPaddress.

What is your base image?
The problem is I can't really provide step by step support for this unless building myself and I have moved off the 32bit XU4 now.

rek79, 2021-06-25:
Base image is armbian buster, the one you recommended.  I completely understand that it will be difficult to troubleshoot given that you have moved on to different hardware.  Maybe you can help me with getting the correct permissions on my external SSD.  I tried starting bitcoind manually while logged in as admin and this is the message I received: 
[image]

rek79, 2021-06-25:
So I switched user to "bitcoin" and started bitcoind manually.  Now it is running but systemd does not know it's running.
[image]

openoms, 2021-06-25:
`sudo killall bitcoind`

to get rid of the daemon and try running the systemd.

debug with tail - ing the bitcoin debug.log (commands in `debug`)
and use `| grep "what you are looking for"`on `sudo journalctl`.

Best to download VScode or a similar IDE to be bale to look at the code you are running.

rek79, 2021-06-25:
I'm familiar with VS Code.  How would I install it on armbiam buster?

openoms, 2021-06-25:
in the terminal it is nano / vi / vim etc

VScode is for your desktop.

rek79, 2021-06-25:
Got it.  Didn't know I could debug from desktop to remote machine with VS Code.  I'll look into it.

rek79, 2021-06-26:
First success!
[image]

openoms, 2021-06-26:
I see you re using the bash terminal of VScode which is great.

You can also download the code to your desktop and open the folder in VScode.
That way you can search for any message or code context easily and know where to edit on the node if need to.

rek79, 2021-07-09:
Found one issue on Line 37 of _bootstrap.sh.  The script fails to retrieve the local IP so I commented out and manually entered for now:
[image]

I'm still having issues getting nginx to start.  I'm going to troubleshoot that next:
Jul  9 10:17:04 odroidxu4 _bootstrap.sh[1179]: sed: -e expression #1, char 40: unknown option to `s'
Jul  9 16:23:47 odroidxu4 nginx[7414]: nginx: [emerg] BIO_new_file("/mnt/hdd/app-data/nginx/tls.cert") failed (SSL: error:02001002:system library:fopen:No such file or directory:fopen('/mnt/hdd/app-data/nginx/tls.cert','r') error:2006D080:BIO routines:BIO_new_file:no such file)

rek79, 2021-07-09:
I think I found an issue.  I've been getting "./_bootstrap.sh: line 615: /home/admin/raspiblitz.log: Permission denied" and "/home/admin/config.scripts/internet.sh: line 222: /mnt/hdd/raspiblitz.conf: Permission denied" messages.  Here are the permissions for files in question:
[image]

[image]

Is that right?

rek79, 2021-07-09:
Hmm.  Should lndAddress in raspiblitz.conf be blank at this point?  Should this be the local IP of my node?  I'm wondering if I need to hardcode the IP address in lnd.setaddress.sh.  I will need to chase down what script calls this one to pass in the ${address} argument.
 
[image]

rek79, 2021-07-10:
So I've definitely got bitcoind running but not lnd.  Here is where I'm currently stuck:
[image]

Debug Logs:
https://termbin.com/b2w3

rek79, 2021-07-19:
The assignment of localIP shows up in a bunch of scripts.  This fails to retrieve the localIP for Odroid HC2 hardware:
`localIP=$(ip addr | grep 'state UP' -A2 | egrep -v 'docker0|veth' | grep 'eth0\|wlan0\|enp0' | tail -n1 | awk '{print $2}' | cut -f1 -d'/')`

This is what `ip addr` returns:
```
admin@odroidxu4:~$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
4: enx001e0631d902: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:1e:06:31:d9:02 brd ff:ff:ff:ff:ff:ff
    inet 192.XXX.X.XXX/XX brd 192.XXX.X.XXX scope global dynamic noprefixroute enx001e0631d902
       valid_lft 5001sec preferred_lft 5001sec
    inet6 xxx:xxx:xx:xxx:xxx:xxx:xxx:xxxx/xx scope global dynamic noprefixroute
       valid_lft 86124sec preferred_lft 14124sec
    inet6 xxx::xxx:xxx:xxx:xxx/xx scope link noprefixroute
       valid_lft forever preferred_lft forever
```
I changed the local IP to "XXX" for security reasons.  It looks like the last grep command needs to be altered to look for "inet"

rek79, 2021-07-19:
I'm fairly certain the localIP variable assignment is screwing up a bunch of scripts.  I've found this to work and also seems much less complex than the grep/egrep:
[image]
`localIP=$(hostname -I | awk '{print $1}')
` 
I wonder if it is worth submitting a pull request to change this in the main branch?

openoms, 2021-07-19:
@rek79 nice find!
Tried it on various machines (RPi4 64bit, Odroid HC1 with Armbian, Debian Desktop and VM) and it indeed works. 
We have a branch in flux (`pre-1.7.1`) which will be merged in the `dev` branch to go forward.
The dev branch will be a good target once we have the current major refactoring done and merged there.

rek79, 2021-07-19:
Glad to help!  It may also be worth putting this in the _commands.sh file so each script can call as needed.  Here are the scripts I found that retrieve localIP in some form or another.
[image]

rek79, 2021-07-19:
This also works:
`localIP=$(ip addr | grep 'state UP' -A2 | egrep -v 'docker0|veth' | grep 'eth0\|wlan0\|enp0\|inet' | tail -n1 | awk '{print $2}' | cut -f1 -d'/')
`
I added "inet" to the end of the third grep command.  It seems to be present in some of the scripts and not in others.

openoms, 2021-07-20:
@rek79 adding `inet` everywhere or using `hostname -I`. Go ahead and make the PR to pre-1.7.1 (until not merged to `dev`).
`_commands.sh` is not a library, only contains shortcuts to be used in the teminal.
Trying to not source too many scripts to others especially not for one function.

openoms, 2021-07-20:
Changing https://github.com/rootzoll/raspiblitz/blob/7ff2ca32f29eb7383ae810dd0d3ffc688863b69c/home.admin/config.scripts/internet.sh#L65
```
localip_LAN=$(ip addr | grep 'state UP' -A2 | egrep -v 'docker0|veth' | egrep -i '(*[eth][0-9]$)' | tail -n1 | awk '{print $2}' | cut -f1 -d'/')
```
to
```
localIP_LAN=$(hostname -I | awk '{print $1}') 
```
worked well.
There is also a one-liner to fix many other instances by adding `\|inet`:
```
grep -rli 'eth0\|wlan0\|enp0' * | xargs -i@ sed -i 's#eth0\|wlan0\|enp0#eth0\|wlan0\|enp0\|inet#g' @
```

rek79, 2021-07-20:
Will work on PR tonight, thanks!

rek79, 2021-07-20:
Submitted some PR's on pre 1.7.1. I'm having trouble bundling them into one patch so I've got separate PR's for each file.  Is there a better way to do it?

rek79, 2021-07-21:
Update:  After fixing the localIP I was still having problems with LND.  Some of the scripts did not have access to write to /mnt/hdd/lnd and had trouble created cert.tls, lnd.config, etc...  This location is owned by bitcoin:bitcoin.  After some head scratching I added "admin" to the "bitcoin" user group and rebooted.  This seemed to solve the issues with LND on restart.  Everything just started working!  lnd.config file copied over, lnd successfully started and initiated wallet, swap file created on HDD, etc... This is what my node currently displays:
[image]

I did run into some issues with some of the bonus scripts not being able to write to the raspiblitz.config file.  I opened a separate issue on that under rootzoll/rapsiblitz #2433 

I'll report back here if I hit any further roadblocks.

rek79, 2021-07-22:
Update:  I discovered that, since permissions were incorrectly set for raspiblitz.conf, the network=bitcoin and chain=main were missing.  I used chmod 777 to fix raspiblitz.conf and manually entered network and chain.  That seemed to fix a weird loop I was encountering when starting SSH session.  I'm now still waiting for sync and will report back once that finishes up.

rek79, 2021-07-23:
Well, looks like I'm up and running with 100% sync!  Even though I'll eventually upgrade from Odroid HC2 to Raspberry Pi 4 this was very helpful.  I'm glad I stuck with it for the learning experience and was even able to merge a PR!  Thank you @openoms for helping me get the Odroid alive and kicking.  Cheers!

[image]
[image]

rek79, 2021-07-23:
Reopened in case you want to confirm successful SDcard image.  Let me know if there is anything I can do to help.

openoms, 2021-07-23:
Great journey!
I use an HC2 for a small LND node for Sphinx and JoinMarket. Uptime is many months.
Another XU4 is serving well as a Joininbox running a pruned node.
The upcoming C-lightning integration might be a good fit as well.

rek79, 2021-07-23:
Sphinx and JoinMarket seem like a good fit.  That will keep me busy for a while, haha.

rek79, 2021-12-22:
Circling back to confirm that v1.7.1 update worked flawlessly using latest Armbian Buster image for Odroid HC2.  All data saved to external HD persisted through the update along with all of my open channels.