raspiblitz/raspiblitz #2347: Running raspiblitz over wifi makes other devices connected to wifi router lose connection.

12cbQLTFMXRnSzktFkuoG3eHoMeFtpTu3S, 2021-06-07:
During  raspiblitz software setup it makes other devices within the local network lose connection to router after a few mins it connection to other devices comes back again.  Not sure if this a normal thing during download.

Devices that are connected are as follows - Mobile Phone, PC on wifi and Tablet. 

A few weeks ago tried to run Umbrel thru LAN and did not experience any disconnection on any of devices. 

Router remains fully operational during these periods.

Current speeds for download and uploads are both  approximately 100Mbps.

rootzoll, 2021-06-14:
Havent seen reports on this from other users, will keep open for others to report on - but maybe an edge case with your router setup or pi hardware. In general LAN connection is recommended over Wifi if possible.

12cbQLTFMXRnSzktFkuoG3eHoMeFtpTu3S, 2021-06-16:
Tried to download Umbrel thru wifi noticed that it's replicates same problem. This seems to be more of a general problem with downloading large files thru wifi even if its not Raspiblitz. 

Can happen to anyone downloading large files thru wifi even persisted with Windows 10 users. Will close comment.

closed 2021-06-16 by 12cbQLTFMXRnSzktFkuoG3eHoMeFtpTu3S