raspiblitz/raspiblitz #2348: Zap Android doesn't connect - SSL handshake aborted

agd-media, 2021-06-07:
While Zeus connects to my node, Zap Android is showing following error:

SSL handshake aborted: ssl=0x7e1a8f9588: I/O error during system call. Connection reset by peer

I tried the bonusxxxx.sh and refreshing the certs. No success.

I wouldn't have this problem if RTL, Thunderhub or Zeus would support lnurl open channel.

NilByte, 2021-06-07:
Zap Android mobile connect was disabled in 1.7.0 menu. 

I had to edit menu to enable it again and then had no trouble connecting Zap Android mobile wallet again.

home.admin/97addMobileWallet.sh

https://github.com/rootzoll/raspiblitz/commit/ba8cd7a62b1e216772c065a2a4757543ae08064a

agd-media, 2021-06-07:
How did you edit the menu? Just deleting the # in 97addMobileWallet.sh won't work.

agd-media, 2021-06-07:
That worked! ZAP connected after like 10 tries.

rootzoll, 2021-06-14:
There were some updates for ZAP wallet - will check for v1.7.1 if we can re-add ZAP by default.

rootzoll, 2021-09-21:
Pushing this up to v1.8.0 (due to time limitations).

rootzoll, 2021-12-09:
Opened `2348zap` branch to test Zap wallet integration again.

rootzoll, 2021-12-09:
OK Zap iOS & Android tests look good again ... adding back to RaspiBlitz SSH connect menu for v1.7.2.
Merging branch into `dev` & ready for final testing.

rootzoll, 2022-02-17:
Thru new tor refactor there is no default tor hidden service anymore for LND Rest or GRPC with v1.7.2RC2 .. need to add them on demand when user wants to connect mobile.

rootzoll, 2022-02-17:
OK Tor service gets now created if needed. keeping for final testing.

rootzoll, 2022-02-20:
OK tested. Ready for relase. Closing issue.

closed 2022-02-20 by rootzoll