raspiblitz/raspiblitz #2349: 0 sat transactions

devotedcorey, 2021-06-07:
I've noticed a few "0 sat" on-chain transactions. The destination of the txn in the picture definitely had sats on it so not sure what's going on.  I don't remember sending that txn either.

[image]

openoms, 2021-06-10:
Check what the command line shows with:

`lncli listchaintxns`

Usage:
```
# List all transactions an address of the wallet was involved in.
# This call will return a list of wallet related transactions that paid
# to an address our wallet controls, or spent utxos that we held. The
# start_height and end_height flags can be used to specify an inclusive
# block range over which to query for transactions. If the end_height is
# less than the start_height, transactions will be queried in reverse.
# To get all transactions until the chain tip, including unconfirmed
# transactions (identifiable with BlockHeight=0), set end_height to -1.
# By default, this call will get all transactions our wallet was involved
# in, including unconfirmed transactions.

$ lncli listchaintxns [command options] [arguments...]

# --start_height value  the block height from which to list transactions, inclusive (default: 0)
# --end_height value    the block height until which to list transactions, inclusive, to get transactions until the chain tip, including unconfirmed, set this value to -1 (default: 0)
```

Also RTL and ThunderHub has options to show the onchain txns.

devotedcorey, 2021-06-10:
Truncated a lot here.  Looking in BTC explorer on my node, the destination address says `is mine = false`. It's also noteworthy that there are 44 inputs, and all are legacy addresses (destination is native segwit).

{
            "tx_hash": "9f5edc......",
            "amount": "0",
            "num_confirmations": 776,
            "block_hash": "00000000.......",
            "block_height": 686288,
            "time_stamp": "1622842936",
            "total_fees": "0",
            "dest_addresses": [
                "bc1q......"
            ],
            "raw_tx_hex": "020000002d6e421......",
            "label": ""
        }

openoms, 2021-06-10:
legacy addresses like 1...? 
your address might have been involved in a dust attack / OP_RETURN maybe?

It is completely fine to not share the txid, but it is difficult to say anything without it.
Close if there is nothing actionable on our side.

devotedcorey, 2021-06-10:
Yes "1..."

TxID 9f5edc1a77a3dcda0a20a1175cf8fb8599a44b5772d39d09daf8a544a19f0a14. I'm just concerned there is something going on with my node that I have no insight into. I'm running joinmarket, but that doesn't look like it's related to that. If none of the addresses are associated with my seed, what could be going on?

openoms, 2021-06-10:
looking at https://kycp.org/#/9f5edc1a77a3dcda0a20a1175cf8fb8599a44b5772d39d09daf8a544a19f0a14
all the legacy addresses belong to CASHAPP. Looks like a consolidation transaction by them.

Did you withdraw from Cash App to your node?

If the destination address does not belong to you not sure why it was picked by your LND. Looks like a bug. Could report to LND: https://github.com/lightningnetwork/lnd/issues

JoinMarket surely has nothing to do with the LND wallet, and neither JoinMarket, nor LND is dealing with legacy addresses.

devotedcorey, 2021-06-10:
Last Tx I did with cashapp was a couple weeks ago. Doesn't look like there are any common addresses between the two transactions.

inpharmaticist, 2021-06-18:
FYI, LND says this can be fixed with an upgrade and rescan.

https://github.com/lightningnetwork/lnd/issues/5384#issuecomment-861815580

openoms, 2022-07-04:
closing for inactivirty

closed 2022-07-04 by openoms