raspiblitz/raspiblitz #2350: Spanish documentation

mosmat, 2021-06-09:
Hey there. I've been over the Spanish translation of the README.md file; [almost there](https://github.com/mosmat/raspiblitz/blob/v1.7/L%C3%89EME.md). Although I'm not sure if something like this has been done already as I think is somehow mentioned at #752. The FAQ.md would be good to have it translated as well, for url's consistency. 

Github freshman, so I'll be figuring. Let me know if a pull request for it would be a proper issue handling.

Regards.

nyxnor, 2021-06-09:
Well, it is good that you have interest, it will be long journey to translate everything.
The path to do great changes, best to start an issue/discussion to provide insights on what you are planning and if it is viable and helpful.
What happens is that every release, the documentation changes, so links can become broken.
You work is helpful, but is it viable for you to update every release? You have to keep that in mind, or ask some spanish friends to help with that too to get some weight off of your shoulder. I can help with spanish translation if you need (not my native language), but I can't be updating everytime the spanish docs, as Im focusing on other things for Blitz at the moment.

If you continue with the spanish docs, I advise you to use stable topics, the ones that won't be changing very often.

mosmat, 2021-06-10:
Yup, I've been noticing the changes whenever the releases; I can grasp the structure of the readme contents and links. 
I think that the original readme is strong consistent when it comes to learn about nodes; as some sort of Hitchhiker's Guide to the Lightning. 
At first I imagined a single line within the original README.md linking to the spanish-translated file, but then I started noticing the links and complexity. 
So far I'm guessing that the faq and the workshop files would be need to be kept updated as well. Not sure about polluting the repo by adding files/folders.
Shiller's my second name, so although I could look for help, maybe just a LÉEME.md file would be doable.

rootzoll, 2021-06-14:
The README and the other files are very big and complex documents that will be hard to keep in sync in others languages. I think for people diving that deep english is the best common language to provide.

But for the v1.8.0 we are working on a WebUI .. which should basically take the other thru the whole setup process - so that the README is just needed for a more indepth info. There we plan to have multi-language support available thru transiflex. It will some more weeks until we are ready - @mosmat if you then like to take the lead there on a spanish translation that would be wonderful.

mosmat, 2021-06-16:
> ... There we plan to have multi-language support available thru transiflex. It will some more weeks until we are ready...

That's cooler and less of a hassle to handle. Then I'll just make sure the links in the Spanish README works for any amigo to have a deep dive; it has a link to the Spanish telegram group anyway.

> then like to take the lead there on a spanish translation that would be wonderful.

Not sure if the lead but of course that I'll be happy to help as I'll keep learning as well. 
Closing this one, then. Cheers!

closed 2021-06-16 by mosmat