raspiblitz/raspiblitz #2351: lnd address are red on clearnet node, but port 9735 is forwarder on a router and open to network

Rudycode, 2021-06-09:
I have been changed my router from tp-link model tl-wr841n to the router amped wirless ac2600, enabled UPnP and enabled BTC UPnP and LND UPnP. These settings on starting my node occur error on over loop on trying to start lnd service
[image]. Then I disabled lnd UPnP and set port forwarding to the raspiblitz port 8333 and 9735 and node successfully started, but lnd node address become red but both ports are opened. 

later I will try to switch back to the tp-link router and see is that a problem on my new router or not.

rootzoll, 2021-06-14:
Maybe try to get feedback on this from the LND Dev Slack directly --> https://dev.lightning.community

If there is anything we can improve on the RaspiBlitz side let us know.

filal1, 2021-07-05:
Hello Rudycode, 
Do you find a solution ?
I have exactly the same problem.
However i have asked to 2 nodes to open a chan with my node and it works (now, 2 chans are opened) but on my screen , the addresses are always in red.
Thank you.

cstenglein, 2023-06-07:
@filal1 @Rudycode is this still relevant? Closing for now, if it's still relevant, please re-open

closed 2023-06-07 by cstenglein