raspiblitz/raspiblitz #2355: upgrade openssh to min 8.2 for fido2 support

canguruhh, 2021-06-17:
it would be great to use openssh-server version 8.2 as it introduced fido2 support by default

originaljeremy, 2021-06-21:
I would love to see fido2 on Raspiblitz

I would be interested in a use case where you could leave the unit on an untrusted network, and lock down ssh unless the security key is inserted into the pi. I'm not aware of anything quite like that in the wild, and all the node stacks have an evil maid weakness - so this would be a big step in the right direction, imho.

canguruhh, 2021-06-21:
login with a local fido2 device is supported by openssh 8.2. not sure if its possible to configure it to require fido2 on the remote side. i guess your use case could be implemented as a PAM script

canguruhh, 2021-06-21:
> I would love to see fido2 on Raspiblitz
>
> I would be interested in a use case where you could leave the unit on an untrusted network, and lock down ssh unless the security key is inserted into the [...]

login with a local fido2 device is supported by openssh 8.2. not sure if its possible to configure it to require fido2 on the remote side. i guess your use case could be implemented as a PAM script

rootzoll, 2021-07-01:
Because we have a lot going on with refactoring I will push this to v1.8.1 to reconsider. But if someone is running tests please report and we may can add this update/feature earlier.