raspiblitz/raspiblitz #2357: Hard-coded codename in Tor sources

bubelov, 2021-06-19:
I found this block of code in `build_sdcard.sh`:

```bash
echo "*** Adding Tor Sources to sources.list ***"
torSourceListAvailable=$(sudo grep -c 'https://deb.torproject.org/torproject.org' /etc/apt/sources.list)
echo "torSourceListAvailable=${torSourceListAvailable}"  
if [ ${torSourceListAvailable} -eq 0 ]; then
  echo "- adding TOR sources ..."
  if [ "${baseimage}" = "raspbian" ] || [ "${baseimage}" = "raspios_arm64" ] || [ "${baseimage}" = "armbian" ] || [ "${baseimage}" = "dietpi" ]; then
    echo "- using https://deb.torproject.org/torproject.org buster"
    echo "deb https://deb.torproject.org/torproject.org buster main" | sudo tee -a /etc/apt/sources.list
    echo "deb-src https://deb.torproject.org/torproject.org buster main" | sudo tee -a /etc/apt/sources.list
  elif [ "${baseimage}" = "ubuntu" ]; then
    echo "- using https://deb.torproject.org/torproject.org focal"
    echo "deb https://deb.torproject.org/torproject.org focal main" | sudo tee -a /etc/apt/sources.list
    echo "deb-src https://deb.torproject.org/torproject.org focal main" | sudo tee -a /etc/apt/sources.list    
  else
    echo "!!! FAIL: No Tor sources for os: ${baseimage}"
    exit 1
  fi
  echo "- OK sources added"
else
  echo "TOR sources are available"
fi
```

Is there a reason for code repetition and hard-coded codename in both branches?

Here is an alternative:

```bash
codename=$(lsb_release --codename --short)
echo "- using https://deb.torproject.org/torproject.org $codename"
echo "deb https://deb.torproject.org/torproject.org $codename main" | sudo tee -a /etc/apt/sources.list
echo "deb-src https://deb.torproject.org/torproject.org $codename main" | sudo tee -a /etc/apt/sources.list    
```

Using repos from wrong OS versions sounds like it can introduce security and compatibility issues.

nyxnor, 2021-06-19:
Responded here https://github.com/rootzoll/raspiblitz/issues/2283#issuecomment-833774093

rootzoll, 2021-07-01:
@bubelov if this answers your question, please close issue. Thx.

bubelov, 2021-07-01:
@rootzoll partly. I didn't find any clear statements on Tor setup code being removed from this script. No code, no problems, indeed. Closing this issue assuming it will be removed. If I got it wrong, feel free to reopen it

closed 2021-07-01 by bubelov

nyxnor, 2021-07-23:
@bubelov https://github.com/rootzoll/raspiblitz/pull/2419/commits/40e6dc20f4eb760c9cef078f5ff637456aa4e446