raspiblitz/raspiblitz #2359: [feature-request] Add charge-lnd by accumulator

DoktorWhat, 2021-06-20:
https://github.com/accumulator/charge-lnd

Description:
This script matches your open Lightning channels against a number of customizable criteria and applies channel fees based on the matching policy.

openoms, 2021-06-20:
Have some manual setup instructions for now (https://t.me/RaspiBlitzFAQ/290):

```
sudo su - bitcoin

git clone https://github.com/accumulator/charge-lnd.git

lncli bakemacaroon offchain:read offchain:write onchain:read info:read --save_to=~/.lnd/data/chain/bitcoin/mainnet/charge-lnd.macaroon

cd charge-lnd 

pip install -U setuptools && pip install -r requirements.txt .

exit

# Then run from admin (same in a cronjob):

sudo -u bitcoin /home/bitcoin/.local/bin/charge-lnd -c /home/bitcoin/charge-lnd/charge.config
```

LupusPluvia, 2021-07-31:
hello unfortunately I don't get the end if I enter sudo -u bitcoin /home/bitcoin/.local/bin/charge-lnd -c /home/bitcoin/charge-lnd/charge.config
  that always comes. Config file not found
What am I doing wrong