raspiblitz/raspiblitz #2361: Vagrant and dev workflow

bubelov, 2021-06-21:
What's the usual dev workflow? Is there any docs or other material on that topic? It looks like the easiest way to spawn a fresh blitz instance is to install `vagrant` and `virtualbox` and then call `vagrant up && vagrant ssh`. It worked like a charm on my Ubuntu PC but later failed with 400 GB error message (looks like the whole thing defaults to mainnet and there are no obvious ways to switch it to testnet/signet and the default disk size is 16 GB).  Any tips or links on that?

Another thing that looked odd is the fact that Vagrant image version is v.1.6.0. Should it match with the blitz version or it's fine to use 1.6.0?

rootzoll, 2021-07-01:
Having a VM/VirtualBox to develop is a future goal - especially when non Hardware relevant features are worked on (you can switch manually to testnet after setup process). But now there is a lot of refactoring going on and VirtualBox is not running on new M1 Apple chips .. so at the moment its back to the developing directly on a RaspberryPi and thats why the vagrant is a bit outdated.

For that you can use the `github` command on the terminal to set the RaspiBlitz to your GitHub repo and branch. And then with the command `patch` you can sync the code on the RaspiBlitz with your repo/branch.

Others like to setup automatic sync of directories with their laptop manually - it depends on your taste.

I would be open to someone reworking the VM setup for development .. for me personally its just on delay to see how the vagrant situation develops on Apple M1 laptops - maybe have to change to something different here.

cstenglein, 2023-06-07:
@bubelov  is this still relevant?

Closing this for now, please re-open if you need any more info.

closed 2023-06-07 by cstenglein