raspiblitz/raspiblitz #2362: Why HTTPS for onion services?

nyxnor, 2021-06-21:
Just lnbits needs it to use the QR code or there is another reason? Needing to accept the self signed certificate does not help security, the hostname is already the pub key we are authenticating to.

Read this https://matt.traudt.xyz/posts/dont-https-your-o44SnkW2/

openoms, 2021-06-21:
it is used because of the IP2Tor service. The `socat` translation would strip the Tor encryption and services would be left with cleartext.
See the SSL stripping attack on Tor users: https://www.zdnet.com/article/a-mysterious-group-has-hijacked-tor-exit-nodes-to-perform-ssl-stripping-attacks/

nyxnor, 2021-06-21:
oh, for the ip2tor, forgot about that. Understood.
Regarding certificates, we need to use sha256, not sha1 in the menu. Just needs to add `-sha256` in every file a service is in and put more width in the dialog box. Closing.

closed 2021-06-21 by nyxnor

nyxnor, 2021-06-25:
Ok, what was bugging me the most is that it was asking to accept the self signed certificate.
But there is an open proposal to make TB verify that https://github.com/alecmuffett/onion-dv-certificate-proposal/blob/master/text/draft-muffett-same-origin-onion-certificates.txt