raspiblitz/raspiblitz #2363: Can't unlock wallet

kotopuz, 2021-06-21:
Dear all,
When I start node I can't unlock wallet. RTL not connecting.
Tried to uninstall  and reinstall RTL feature, Tried same with Thunderhub - nothing works. Tried to remove funds - not works.
Tried to reinstall raspiblitz, same.
See log 
https://termbin.com/bg98

Thanks!

openoms, 2021-06-21:
your lnd database (channel.db) does not look healthy:
```
21-06-21 13:50:51.766 [INF] CHDB: Checking for schema update: latest_version=20, db_version=20
2021-06-21 13:50:51.768 [INF] LTND: Database now open (time_to_open=360.584696ms)!
2021-06-21 13:50:52.118 [INF] RPCS: Password RPC server listening on 0.0.0.0:10009
2021-06-21 13:50:52.122 [INF] RPCS: Password gRPC proxy started at 0.0.0.0:8080
2021-06-21 13:50:52.122 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2021-06-21 13:54:41.216 [INF] LNWL: Opened wallet
2021-06-21 13:56:44.632 [ERR] LNWL: Failed to open database: timeout
2021-06-21 14:01:30.001 [ERR] LNWL: Failed to open database: timeout
2021-06-21 14:40:32.253 [ERR] LNWL: Failed to open database: timeout
2021-06-21 15:00:21.867 [ERR] LNWL: Failed to open database: timeout
2021-06-21 15:13:15.818 [ERR] LNWL: Failed to open database: timeout
2021-06-21 15:55:40.724 [ERR] LNWL: Failed to open database: timeout
2021-06-21 16:21:52.086 [ERR] LNWL: Failed to open database: timeout
2021-06-21 16:24:04.120 [ERR] LNWL: Failed to open database: timeout
```

Are you on raspiblitz v1.7.0?

Does `uname -m` show `armv8` (64 bit)?

kotopuz, 2021-06-21:
Yes. 1.7.0
┌─────────────────────┤  LND ERROR  ├──────────────────────┐                                               ############################
Calling: lncli unlock
Please re-enter Password C:
admin@raspberrypi:~ $ uname -m
aarch64
admin@raspberrypi:~ $

openoms, 2021-06-21:
if you have no funds just need to reset the wallet from the REPAIR menu after typing wallet

of there are channels will probably need to restore  from SEED + SCB. See the FAQ for the options.

kotopuz, 2021-06-22:
Performed LND reset and restored that from backup file.
Now I got"
"****************************************
* The LND service is running with error.
****************************************
If you just started some config/setup, this might be OK.

5 RESTARTS DETECTED - LND might be in a error loop
1624356991 ERROR: [lncli] could not load global options: unable to read macaroon path (check the network setting!): open /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory
"

kotopuz, 2021-06-22:
here is log
https://termbin.com/sdt8

openoms, 2021-06-23:
as in the lnd logs:
```
2021-06-22 11:29:41.470 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
```

Try:
`lncli unlock`

or if there is no wallet run:
`./70initLND.sh`

kotopuz, 2021-06-24:
NEW INSTALL. New SD new HDD and restore from lnd-rescue file.
Same error, wallet is not unlocking!
After I try lncli unlock it says it's unlocked now,  but when run menu to see info screen I got
error-28
Can you make please a working version of raspiblitz ?
I am struggling to recover funds for a month already.

openoms, 2021-06-24:
The -28 error suggests that your bitcoind is syncing.

Look at debug.

Try a restart for the menu to pick up a status correctly.

kotopuz, 2021-07-01:
Hello, menu restart not help. I've deleted all data from HDD, restored from rescue file, resynced blockchain.
Now I see "LND   Not Started | Not Ready Yet".  When I run INFO from menu it always asking LND password. lncli unlock with password says it's unlocked and again with info it's asking password. The nit goes back to menu, then I run info again and it says LND not started! AAAAA.
Tried restart, shutdown - all the same.
I've tried reflash SD, erased HDD why I can't make is work !

kotopuz, 2021-07-01:
here is the log https://termbin.com/zmw7
Also tried to unlock wallet from RTL, tried to restart lnd service from cmd. Nothing works.

openoms, 2021-07-01:
Can you list your hardware details? 
There are some permission errors in your log I haven't seen before. Could be a failing disk. 

If you have no funds in channels (only onchain) restoring your LND (aezeed format) seed (24 words) in Blue Wallet would give access to your funds.
If there are funds in channels you would need to restore from the seed ans the channel.backup file as described in the FAQ. 

Afterwards I'd would take the recommended hardware parts, format the disk on the desktop and flash the latest image to start new.

kotopuz, 2021-07-02:
All in enclosure and I hope I can get a hardware details from some command. Is there is HW report available ?
It's 1TB HDD connected to RPI4b 2GB via USB adaptor.
Main problem is that it has funds onchain and offchain what I try to recover. Previous version was working fine, such problems started with new one only. I've started new already with format and new flash image, but I am still getting same errors. CLI lncli unlock says it's unlocking wallet. Is there is command to start manually lnd or force start ? Thunderhub connected once and not able to connect to node since then, RTL also not working.

openoms, 2021-07-02:
It is most likely that you channel.db (part of the rescue or migration file) was currupted on the disk and cannot be restored.

Proceed with restoring from the Seed and Static Channel Backup: https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#2-recover-from-wallet-seed

This will force close you channels with the online peers and  returns the funds onchain.

kotopuz, 2021-07-05:
Made channel backup to USB. and made restore with channel backup. No helped. Then started new LND wallet waited for sync and restored again and WORKED! All remaining channels closed and in 2 days I got transactions confirmed. All funds taken out immidiately from LND. Few month trying different things, downloading blockchain, checking sectors in HDD, changing SSD, rescue with many options, trying chain-tools to recover lost channels funds and learning how to use CLI commands from stuck or lost channels! IT was disaster. Thanks again to everyone for your support, but now I am our from Lightning network. I've read multiple times about it as warnings to be careful with it and that it's still no adopted properly and that it's costly and risky with reward. It's good to be enthusiast until you loose real money you earn soo hard. Will wait until LND configs will be bullet proof with built in mechanisms to protect funds and easy ways to restore. Unfortunately it's not the case. Raspberry is NOT safe device, to use funds and make some earning you need PC (better virtual machine) with RAID disks, UPS and auto backups configured to minimize risk of loose funds. Failing to implement even those basics is a high risk and shooting in you own leg in long term. Be careful node operator and stay safe! Overall it's getting better and I think in few years we'll get a reliable config and maybe RPI 5 or 6 will have built in RAID controllers.
Last thing for developers.
Please try to find a way to implement HW redundancy and fall back options as:
1. RAID HDD/SDD for raspiblitz config. SD card is not reliable.
2. RAID HDD/SDD for blockchain. Even crappy HDDs in Raid would be fine as one fails you bring another one and you're done!

Suggest some cheap external raid controller (there are some in Chinese ) market for 1 and 2 above.
Make UPS interface to shutdown RPI if UPS send contact or signal of mains failure. Suggest some check UPSs for RPI with tested solution.

and Finally make backups automatic! Dropbox config looks ugly (it works but ugly). Make all backups as rescue, channels, daily/weekly etc. Avoid closing channels on restore or find way to close channels as it's lost funds!

Have fun! Bye Raspiblitz!

openoms, 2021-07-05:
Glad you have the funds returned now and thanks for the suggestions.
Most of them are implemented or discussed already:

Multiple options to backup SCB-s:
https://github.com/rootzoll/raspiblitz/blob/v1.7/README.md#backup-for-on-chain---channel-funds

UPS recommendation and support: https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#how-do-i-connect-a-ups-to-the-raspiblitz

A simple mdadm RAID is far from being a silver bullet unfortunately since corrupted data can be replicated easily and no solution for the cold shutdowns.

A checksumming, snapshot-able (copy-on-write) filesystem like ZFS or BTRFS is better and we are experimenting with it: https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#why-use-btrfs-on-raspiblitz

Ultimately you are right, the Raspberry Pi is only a cheap standardised option to start and significant funds should be moved to a more robust setup like:
https://github.com/seth586/guides/blob/master/FreeNAS/bitcoin/README.md
https://twitter.com/openoms/status/1407377844024528905

More reading:
https://github.com/lightningnetwork/lnd/blob/master/docs/safety.md
https://github.com/lightningnetwork/lnd/blob/master/docs/recovery.md

https://lightning.readthedocs.io/BACKUP.html

closed 2021-07-05 by openoms