raspiblitz/raspiblitz #2365: Feature Request: Incognito screen display

maestro32, 2021-06-22:
It would be cool to have an option to switch to have a more incognito screen display option. Like perhaps just that cool lightning bolt logo, or the Bitcoin logo without any of the other information such as alias, pub key, channel count/capacity, etc. My main goal is just not to advertise to house guests how much I have on my node if they happen to see and understand the information on the display.

Of course, the simple solution would be to just take the screen off.. but I really like the screen.

openoms, 2021-06-22:
A simple if statement can achieve this.
Play around with the `00infoBlitz.sh` on your node as you like.
PR-s are welcome.

closed 2021-06-23 by maestro32

rootzoll, 2021-06-23:
@fusion44 maybe something to keep in mind for the new TouchUI

fusion44, 2021-06-23:
Yes, I have already a PIN input screen in place. 
You'll be able to hide views behind a PIN.