raspiblitz/raspiblitz #2366: Suez - Tool for pretty printing and optimizing Lightning Network channels for LND and CLN

openoms, 2021-06-22:
https://github.com/prusnak/suez works for both lnd and cln on all networks.
Would be a good replacment for the `lnchannels` command which is only implemented for lnd on mainnet anyway.

tested:
* LND testnet
[image]
* CLN testnet
[image]

To install: 
```
sudo su bitcoin

# https://github.com/python-poetry/poetry#installation
curl -sSL https://raw.githubusercontent.com/python-poetry/poetry/master/install-poetry.py | python -

# You can uninstall at any time by executing this script with the --uninstall option,
# and these changes will be reverted.

# add 'export PATH="/home/bitcoin/.local/bin:$PATH"' or use full path

# https://github.com/prusnak/suez
git clone https://github.com/prusnak/suez.git
cd suez
/home/bitcoin/.local/bin/poetry install

exit
```
Usage:
```
sudo su bitcoin

/home/bitcoin/.local/bin/poetry run ./suez --help

Usage: suez [OPTIONS]

Options:
  --base-fee INTEGER          Set base fee.
  --fee-rate INTEGER          Set fee rate.
  --fee-spread FLOAT          Fee spread.
  --time-lock-delta INTEGER   Set time lock delta.
  --client [lnd|c-lightning]  Type of LN client.
  --client-args TEXT          Extra arguments to pass to client RPC.
  --show-remote-fees          Show (estimate of) remote fees.
  --help                      Show this message and exit.

# does not run with zero channels - check

# lnd
/home/bitcoin/.local/bin/poetry run ./suez 

# cln
/home/bitcoin/.local/bin/poetry ./suez --client=c-lightning --client-args=--conf=/home/bitcoin/.lightning/tconfig

exit
```

bubelov, 2021-06-22:
Tried to install it and change fees, worked fine with mainnet lnd and a bunch of channels

xanoni, 2021-07-03:
Running into trouble with this ... which lnd, python, and poetry versions was this tested with?

See: https://github.com/prusnak/suez/issues/23

xanoni, 2021-07-03:
See my referenced ticket. Was just fixed in master and it's now working for me as of 67620b2601ce35e8ec783079c9d4a1baf1b665a3

openoms, 2021-07-05:
Works well with remote nodes also an example with an LND node on the LAN (installed under the `admin` user):
```
cd ./command-centre/suez
/home/admin/.local/bin/poetry run ./suez --client-args=--rpcserver=REMOTE_NODE_IP:10009 --client-args=--tlscertpath=./../tls.cert --client-args=--macaroonpath=./../admin.macaroon
```
Related: https://github.com/rootzoll/raspiblitz/issues/2392

openoms, 2021-07-17:
install under any user (including admin):
```
curl -sSL https://raw.githubusercontent.com/python-poetry/poetry/master/install-poetry.py | python -
git clone https://github.com/prusnak/suez.git
cd suez
../.local/bin/poetry install
```

Usage (if there is a local copy or symlink of the tls.cert and the admin.macaroon for LND):
```
cd suez && ../.local/bin/poetry run ./suez
```

xanoni, 2021-07-17:
I found it useful to just run this in a screen in an endless loop ... not sure if that's how it's supposed to be done, but it works for me... something like (reconstructed the below from memory, so plz check):

```bash
#! /usr/bin/env bash

BASE_FEE=1000
FEE_RATE=50
FEE_SPREAD=2
TIME_LOCK_DELTA=$((6*24*75/100))
INTERVAL=$((60*30))

cd ~/suez

while true; do
    date
    poetry run ./suez \
        --base-fee "${BASE_FEE}" --fee-rate "${FEE_RATE}" --fee-spread "${FEE_SPREAD}" \
        --time-lock-delta "${TIME_LOCK_DELTA}" \
        --show-remote-fees --show-scores
    sleep "${INTERVAL}"
done
```

I haven't fully thought through if `--show-remote-fees` and `--show-scores` have any privacy implications. Maybe, maybe not.

openoms, 2021-07-17:
@xanoni nice solution, but I find charge-lnd to be the best to manage fees by different policies: https://github.com/rootzoll/raspiblitz/issues/2359

Planning to implement Suez as a visualization tool with the option to use the CLI.

xanoni, 2021-07-17:
> charge-lnd

Cool I'll have a look!

xanoni, 2021-07-19:
There are a lot of useful CLI tools for lnd, such as `bos`, `suez`, `rebalance-lnd`, `lnmanage`, `lntop`, etc. ... but there's also quite some overlap in functionality between them when it comes to things like fee configuration or rebalancing.

How do you handle this in RaspiBlitz? Do you recommend one tool for each task? I don't have a Blitz myself, but would be curious because I get confused at times with all these options :)

rootzoll, 2021-09-20:
Suez will be part of v1.7.1rc2. Putting to final testing.

xanoni, 2021-09-21:
Cool. I'm still using `suez` as my primary fee management tool. Have a cronjob and auto-run it every X hours. In the same script I then overwrite any channels that I want to give preferential treatment (or the opposite ;-).

Based on this thread I understand that there are better solutions (e.g., `charge-lnd`), but `suez` is working well enough for me so have felt no pressure to switch so far.